rogerclarkmelbourne/Arduino_STM32 #332: Add a new RTC example

csnol, 2017-08-28:
[https://github.com/csnol/STM32-Examples](url)

It's a part of my internet clock without the codes about LCD and ESP8266.

If useful, my pleasure!

rogerclarkmelbourne, 2017-08-29:
Thanks

BTW. The link in your issue, did not work, but found it was 

https://github.com/csnol/STM32-Examples

I'll see if I have time to add it to the other examples (as we have a folder for examples)

RickKimball, 2017-08-29:
Can we let this thing have some flight time before blindly adding it to the core libraries?

csnol, 2017-08-29:
Hi @rogerclarkmelbourne  and @RickKimball 
I just wanna add the Timelib library to STM32F1 system. 
because the RTClock.h hasn't  include year(), month() day() hour() minute() and second() function in the STM32F1 of this repo. (I don't know why them be included in  RTClock.h in  STM32F4)

On the other hand,  I can't sure whether  or not its license  can be used in this Arduino_STM32 repo.
Its  license is  GNU Lesser General Public License,  NOT MIT of Arduino_STM32

```
  time.c - low level time and date functions
  Copyright (c) Michael Margolis 2009-2014

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
  ```
I really don't understand about these freeware license.

Thanks guys!
BR.
C.S Nol.

stevstrong, 2017-08-29:
You may have a look to the RTC lib for F4, I have reworked it currently to be independent from any external time lib: https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F4/libraries/RTClock/src

csnol, 2017-08-29:
Thanks @stevstrong .
I've gotten it. very nice codes . and I'll try to copy some lines for F103.

csnol, 2017-08-29:
I've renew the RTColck library, Don't dependents external time lib anymore.
and  pull requested to master repo. 

### Thanks @stevstrong  , Your RTClock library of STMF4 give me a great help.

closed 2017-09-23 by csnol