rogerclarkmelbourne/Arduino_STM32 #335: add max number of pins

aster94, 2017-08-30:
seems that in arduino they added some define for the max number of pins in pins_arduino.h, I just found a library that had this defined
I don't think that it would be a problem for anyone but maybe if you want to keep it as compatible as possible it could be added

example for C series, i added it in variant.h
#define NUM_DIGITAL_PINS            34
#define NUM_ANALOG_INPUTS      10

rogerclarkmelbourne, 2017-08-30:
This could be added, but I'm not sure if it would cause problems

is NUM_DIGITAL_PINS on AVR the number of pins labelled Dx e.g. D0 - D13 ?  or does it include the A0 - A5 pins which can also be used as digital

On STM32 all the pins that support analog input are primarily digital pins

aster94, 2017-08-30:
I checked and it includes it, for this reason i wrote 34 as digital pins
i have the number of analog and digital pins for all the package (T, C, R, V, Z) already made for other reason, if you need it tell me and i copy it here

rogerclarkmelbourne, 2018-05-23:
Looks like the Arduino SAMD core has these added , in 2014

https://github.com/arduino/ArduinoCore-samd/search?utf8=%E2%9C%93&q=NUM_DIGITAL_PINS&type=

So we should possibly attempt to add them

Though I think they are potentially confusing because of the STM32 architecture

stevstrong, 2018-05-23:
I wonder what is the real use case for having this define.

fpistm, 2018-05-23:
It is used in several libraries.
Ex in firmata:
https://github.com/firmata/arduino/blob/master/Boards.h

stevstrong, 2018-05-23:
I still cannot see any benefit of it.

Even if you would define this number for each board, you can have less pins available than this, dependent on the application, so that you will have to re-define it again.

But this is only my opinion.

fpistm, 2018-05-23:
Other classic example for NUM_ANALOG_INPUTS is to use it to iterate over analog:

```
for(i=0; i < NUM_ANALOG_INPUTS ; i++)
analogRead(i);
```

But I think this will not work on this core

rogerclarkmelbourne, 2018-05-23:
OK.

I agree its pointless defining these.

I'll close this issue as "Won't support"

closed 2018-05-23 by rogerclarkmelbourne