rogerclarkmelbourne/Arduino_STM32 #336: OLED_I2C lib sketch size error

AcThPaU, 2017-08-30:
The OLED_I2C lib port is not working because HW_STM32.h still uses HardWire.h.

I changed it to Wire.h and all examples excess 64K.

stevstrong, 2017-08-30:
Please give us more information.
Where is the OLED lib?
Which example have you tested?

AcThPaU, 2017-08-30:
The OLED_I2C lib is one of the default lib in this project, made by RinkyDinkElectronics. I tried all the examples (3d_cube, bitmap.. etc). All of them excess 64k compiled.

stevstrong, 2017-08-30:
Do you really expect from us to help you, to grab all information that you are not willing to share with us?
Is it really sooo hard to attach some links?

AcThPaU, 2017-08-30:
The lib is two years old, ported by [@Madias ](https://github.com/Madias)
https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/OLED_I2C

This is the file I changed:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/OLED_I2C/hardware/arm/HW_STM32.h

As I understand the project now use hardware I2C as default and HardWire.h is no longer available. 

I don't understand why it gets so big when using Wire.h. I remember OLED runs fine with ATmega328, which has only 32K flash.

AdoHaha, 2018-01-25:
I guess I have the same problem. I get: 
2018.1.24/libraries/OLED_I2C/hardware/arm/HW_STM32.h:22:7: error: 'HWIRE' was not declared in this scope
       HWIRE.begin();

error when running OLED_I2C_3D_Cube example from Examples/OLED_I2C/Arduino, also with other examples from this library.
Arduino is set up for standard Blue Pill

rogerclarkmelbourne, 2018-05-10:
Please discuss this on the forum.

closed 2018-05-10 by rogerclarkmelbourne