rogerclarkmelbourne/Arduino_STM32 #341: need to include usb_hid_device.h to boards_setup.cpp in addMidiHID branch

arpruss, 2017-09-19:
In the addMidiHID branch, you need `#include <usb_hid_device.h>` at the top of variants/generic_stm32f103c/wirish/boards_setup.cpp . There may be similar issues in some of the other variants, but that's the one that I tried

rogerclarkmelbourne, 2017-09-19:
OK.

Thanks.

That branch is somewhat out of date, and I should first try to merge the current master

rogerclarkmelbourne, 2017-09-21:
Can you post some code which demonstrates what the problem is e.g. a small amount of sketch code.

arpruss, 2017-09-21:
The Joystick example shouldn't compile without this change. You'll get an error that HID is undefined from the compiler.

rogerclarkmelbourne, 2017-09-21:
OK... 
Will do... (when I get time)

BTW.

The main reason that this isn't rolled into the main / master branch, is that the USB upload relies on USB Serial to trigger the core to reboot the MCU and hence run the bootloader.

When its in HID etc mode, its not possible to send a command to reboot via Serial

I guess we could add another menu option which defaults to USB Serial, but the Arduino IDE menu'ing system does not cascade and we have other upload options apart from USB , which makes things complicated and confusing

feurig, 2017-09-21:
This is one of the reasons why the pluggable usb  creates a composite device. That way the serial never really goes away..

rogerclarkmelbourne, 2017-09-21:
@fergul 

Yes.

But composite devices of different types don't work very well on Windows versions prior to W10
(I'm not even sure W10 supports it very well either)

FYI. 
The BMP looks like it has a composite device for DFU and Serial, 
But the DFU device is not really running, i.e you can't upload via DFU even though in the Windows device manager, it looks like there is a DFU device.

If you try to connect via DFU Util, it will report an error.

feurig, 2017-09-21:
Seems like it would be a no brainer to use class independent vendor commands to endpoint 0 for things like reseting and running the boot loader.

May have to jump back in here.

rogerclarkmelbourne, 2017-09-21:
@fergul 

No idea.

I'm not a USB specialist.  

I know a few people have had experimented with composite devices and also Serial devices that don't need a special driver (it was probably either danieleff or ddrown  but I can't be sure)

But no one seems to have managed to get either a composite device or a driverless serial, to work with W7 or W8.  I think driverless serial works on W10, but as most techies (me included) have an aversion to W10, for good reasons, those changes have never made it into the mainstream

arpruss, 2017-10-15:
And while one is at it, one needs to add `#include <usb_midi.h>` to boards_setup.cpp

arpruss, 2017-10-16:
The issue of having to reset manually isn't a big deal. Once the upload messages show up in the console, I just unplug and re-plug the USB cable, and all is well. 

The branch does have a USB menu, and if you choose Serial, all works well.

rogerclarkmelbourne, 2017-10-16:
@arpruss 

Yes. Thats the problem if the board is not enumerating as USB serial.

Perhaps there is a clever way to send a reset via the keyboard driver (HID) driver.
I though it was possible to send to a HID device, but I have no experience on how to do this

Though there does seem to be some code on the web that claims to be able to send to HID https://github.com/signal11/hidapi

arpruss, 2017-10-16:
Another option is just to print a message to press the reset button.
The Digispark core does something like that.

rogerclarkmelbourne, 2017-10-16:
Unfortunately people don't always have verbose logging turned on, so I don't know they would see a message even if one was printed by a dummy upload script (whose only purpose was to print a message)

arpruss, 2017-10-16:
I just turned off all the messages I could (verbose compile, verbose update, warnings), and I still got this: 

> Sketch uses 21524 bytes (32%) of program storage space. Maximum is 65536 bytes.
Global variables use 3344 bytes (16%) of dynamic memory, leaving 17136 bytes for local variables. Maximum is 20480 bytes.
maple_loader v0.1
Resetting to bootloader via DTR pulse
Reset via USB Serial Failed! Did you select the right serial port?
Searching for DFU device [1EAF:0003]...
Assuming the board is in perpetual bootloader mode and continuing to attempt dfu programming...

>dfu-util - (C) 2007-2008 by OpenMoko Inc.
Couldn't find the DFU device: [1EAF:0003]
This program is Free Software and has ABSOLUTELY NO WARRANTY

I assume one could add to this a message (maybe to stderr) that says: "YOU MAY NEED TO PRESS THE RESET SWITCH OR UNPLUG AND REPLUG YOUR BOARD NOW" before the maple_loader message.

rogerclarkmelbourne, 2017-10-16:
Cool

Which OS ?? Windows ??

arpruss, 2017-10-16:
all of them. It's in the big pull request I sent.

stevstrong, 2019-09-20:
Can this be closed?

rogerclarkmelbourne, 2020-05-18:
Closing because of no activity, and because there is a pending PR for a big change to the USB library

closed 2020-05-18 by rogerclarkmelbourne