rogerclarkmelbourne/Arduino_STM32 #343: delete tools/win/cyg*.dll

arpruss, 2017-09-20:
I think none of the executables in tools/win actually depend on the cygwin dlls, and if they are deleted then there is no need to include a license for them, which is nice.

rogerclarkmelbourne, 2017-09-20:
Good point

They were legacy items I must have copied there for some other reason...

I'll try deleting them and confirm it still works OK.

arpruss, 2017-09-20:
I grepped the executables for the dll names and came up blank in the branch I'm using (addMidiHID), if that helps.

rogerclarkmelbourne, 2017-09-20:
OK.

BTW.

The addmidihid branch is quite a long way behind the master, you may want to try merging the master and see if the usb midi hid stuff still works, as there are numerous bugfixes and enhancements in the master

Unfortunately, I don't have time to maintain any of the branches, I just about keep up with the master

arpruss, 2017-09-21:
I tried merging. Building worked, but when I uploaded to the device (a black pill) I got an unrecognized USB device error in Windows. So it's not as straightforward as one might hope. :-(

rogerclarkmelbourne, 2017-09-21:
OK.

Thanks for trying...

arpruss, 2017-09-21:
And the branch works for my application so far (a GameCube controller to USB adapter).

rogerclarkmelbourne, 2017-09-21:
Issue should be fixed by commit

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/f8d28b8663a53fa37f75c2f20ed0aebc81c83fba

closed 2017-09-21 by rogerclarkmelbourne