rogerclarkmelbourne/Arduino_STM32 #344: SPI library const in signature for write() and transfer()

greiman, 2017-09-21:
The const qualifier was recently removed from the signature for some SPI member functions and was missing from others.

It would be nice if const was added or restored to the first argument of member functions like these:

```
    void write(void * buffer, uint32 length);
    uint8 dmaTransfer(void * transmitBuf, void * receiveBuf, uint16 length);
    void dmaTransferSet(void *transmitBuf, void *receiveBuf);
    uint8 dmaSend(void * transmitBuf, uint16 length, bool minc = 1);
    void dmaSendSet(void * transmitBuf, bool minc);
    uint8 dmaSendAsync(void * transmitBuf, uint16 length, bool minc = 1);
```

Sorry if this is a bit picky but I have started using code analyzers like cpplint, CppCat, Cppcheck to check my code so I find problems like this in the underlying libraries.

Code checkers don't like it if you cast away const in a function call.

rogerclarkmelbourne, 2017-09-21:
Can you do a PR for this, so everyone can test your proposed change

victorpv, 2018-04-12:
Wasn't this merged already?I remember seeing a PR from Bill for the SPI library merged a while back, and the code show the const qualifier:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/SPI/src/SPI.cpp#L394

I guess this issue can be closed as resolved.

rogerclarkmelbourne, 2018-04-12:
@stevstrong 

We could try changing them, but it may fail to compile

stevstrong, 2019-09-16:
Seems to be already solved.

closed 2019-09-16 by stevstrong