rogerclarkmelbourne/Arduino_STM32 #356: Time zone in RTC

staticmem, 2017-10-15:
In the example sketch "STM32F103C8 Blue Pill" I can't set the timezone to 10.5 hours as it is an integer.  I change it to #define timezone 10.5 and it compiles ok but does not show the correct time when I set the epox time.  My work around is tt+=1800 after the epox is entered to add 30 minutes.  Can't say I'm much of a c++ programmer so not sure what the issue is in the library or if it's me.

rogerclarkmelbourne, 2017-10-15:
Which example, there are hundreds of examples (Albeit not all deal with timezones)

staticmem, 2017-10-15:
arduino-1.8.4/hardware/Arduino_STM32-master/STM32F1/libraries/RTClock/examples/BluePill-RTClock-test/BluePill-RTClock-test.ino

stevstrong, 2017-10-15:
Hm, i didn't know that there are time zones with non-integer hours...
Which country is this?

staticmem, 2017-10-15:
South Australia and Iraq, both are on this +9:30 or +10:30 depending on DST.  (It's a problem also with some consumer equipment not supporting it too.)

stevstrong, 2017-10-15:
I would suggest to adjust the time by adding the difference seconds to the epoch time: (9*60+30)*60

rogerclarkmelbourne, 2017-10-15:
(LOL)

I was about to most that South Australia is 30 mins behind us in Melbourne.

AFIK this is not the only place in the world that is on a 30 min boundary

staticmem, 2017-10-15:
Could you have it so 0-23 could still be used but if >= 30 then assume minutes.  That way no need for a new function?

rogerclarkmelbourne, 2017-10-15:
I think that could cause a lot of confusion.

Better just to change it to accept floats.

I think int's should get automatically cast to floats, but if not we could overload the function with one signature which takes and int and another that takes a float

However the internal operation of the class will also need to be changed to handle this

@staticmem 

You may as well have a go at updating the library and posting your solution, otherwise you may have to wait indefinitely for the original author (who wasnt me), so add the functionality you need

staticmem, 2017-10-15:
I'll just use a work around for the time being.  I'm a "C" programmer and don't know much about C++ Classes and don't have the time currently (pardon the pun) to study up on it to look at a fix.

rogerclarkmelbourne, 2017-10-15:
Which function causes the error?

PS. In c++ you can have

void funcA(int a)
{
// SOME CODE
}

and

void funcA(float a)
{
// SOME CODE
}

And the compiler works out which one to call based on whether you pass an int or float etc

However this can cause confusion / problems when the compiler has to cast from something else to either int or float

staticmem, 2017-10-15:
The Example has:
`      rtclock.TimeZone(tt, timezone);    //adjust to your local date
      rtclock.setTime(rtclock.TimeZone(tt, timezone));`
I think that first line is not doing anything as it happens again in the next line.

PS: I tried your overload example but I had to change float to double for it to compile without error. With float the error is:
error: call of overloaded 'funcA(double)' is ambiguous
   funcA(456.7);

rogerclarkmelbourne, 2017-10-15:
FuncA was just a general example...

Did you try changing 

`time_t` TimeZone(time_t t, int TZ) { return ( t + (TZ * SECS_PER_HOUR)); } // usage: localtime = TimeZone(UnixTime, 8);

to 

`time_t` TimeZone(time_t t, float TZ) { return ( t + (TZ * SECS_PER_HOUR)); } // usage: localtime = TimeZone(UnixTime, 8);

or if that gives a compile error, you could try 

`time_t` TimeZone(time_t t, double TZ) { return ( t + (TZ * SECS_PER_HOUR)); } // usage: localtime = TimeZone(UnixTime, 8);

As far as I can see the time_t type is just an integer; hence as SEC_PER_HOUR is divisible by 2 (i.e time 0.5) its still going to result in a value which is a valid integer

staticmem, 2017-10-16:
RTClock.h: With just the one function (no overloading) the 'float' compiles ok.  To have both functions I needed to change float to double as per your example.  Built example again and #define timezone 10.5 and set the Epox time and it is now working.

rogerclarkmelbourne, 2017-10-16:
I'd just change the function so that it accepts a float or double instead of an int, because like you say the version of the function which takes an int is superseded by the new version.

I'm not sure why it won't accept a value of 10.5 as a float, but making the function take a double is not a big problem, it will be a tiny bit slower than a double, but I suspect thats not going to make any difference to 99.999% of the people who would use that function

staticmem, 2017-10-16:
Are you going to make the changes to your branch and/or let the author know?

rogerclarkmelbourne, 2017-10-16:
@csnol  is the last person who sent an update

Hopefully he can comment on this change

csnol, 2017-10-18:
sorry, I don't know there are half-hour timezone.
but I don't think it is a good idea that define float/double TZ.
Maybe I'll add a flag for this function.

staticmem, 2017-10-18:
Hmm, ok it's your project.  Don't or Didn't know?, yes there are time zones that use multiples of 1/2 an hour.  Why don't you think it's a good idea?

csnol, 2017-10-18:
How about that?  

``` c++
// Usage:  1.  localtime = TimeZone(UnixTime, 9, 1)  means SAT +09:30 TimeZone; 
//              2.  localtime = TimeZone(UnixTime, -3, 1)  means NST,NFT -03:30 TimeZone;
//              3.  localtime = TimeZone(UnixTime, 8, 0)  means CCT +08:00 TimeZone, 

time_t TimeZone(time_t t, int TZ, bool HFZ)  {    // HFZ : Half-hour TimeZone flag
if(HF) {
if(TZ > 0 )
 return ( t + (TZ *  SECS_PER_HOUR)+1800); 
else
 return ( t + (TZ *  SECS_PER_HOUR-1800); 
}
else
return ( t + (TZ * SECS_PER_HOUR));
}
// Keep it
time_t TimeZone(time_t t, int TZ) { return ( t + (TZ * SECS_PER_HOUR)); } // usage: localtime = TimeZone(UnixTime, 8);
```

staticmem, 2017-10-18:
Seems awfully complicated to me but maybe I'm missing something here?  What you seem to be proposing means you have to test if time zone in hours is evenly divisible by 1 and set the flag if not.  Now you need to add extra code in the application to decide to set flag to 0 or 1 for a program that simply asks for the time.

csnol, 2017-10-18:
@staticmem At first. Thanks for your issue.
 You know, So many ways can realize this function. one of these  is " tt += 1800" you said.
this function needed to return a long integer time_t. 
it is difficult be controlled and increase the risk If use float/double TZ, (like 10.3 or 3.14 etc.)
up to now I think  "int + flag" is safe way.  
@rogerclarkmelbourne  @stevstrong  Have any suggestions ?

csnol, 2017-10-18:
OK, I accept this function.  It looks simplest way. 
``` c++
time_t TimeZone(time_t t, double TZ) { return ( t + (TZ * SECS_PER_HOUR)); } // usage: localtime = TimeZone(UnixTime, 8);
```

stevstrong, 2017-10-18:
For me, it seems to be an overkill to use double or float variables just because of a half-hour.
I personally support simply solutions which avoid float/double usage.

sweetlilmre, 2017-10-18:
Unfortunately there are also 45 minute offset zones:
https://www.timeanddate.com/time/time-zones-interesting.html
I also agree that a double or float 'feels' wrong. The 'proper' way would
be to have a list of allowable zones and a way to map them to the offset
but that in turn could be hugely intense in terms of flash / RAM.

What about something like a macro that takes hours and minutes and converts
into seconds with the function only accepting seconds and performing a
sanity check for whole hours, 30 minutes or 45 minute offsets?
e.g. #define OFFSET_TO_SECONDS(hours, minutes) ((hours * 60 * 60) +
(minutes*60))

Check is something like:

int mod = ts % 3600;
if (mod != 0 || mod != 1800 || mod != 2700)
{
  // invalid offset
}

rogerclarkmelbourne, 2017-10-18:
As far as I can see there are 3 options

Loads of constants which define every possible timezone, and convert to the underlying datatype, which I think is int32 or unit32

or

Specify hours and minutes rather than just the Hour component

or

Allow floating point

If we go with the hours,minutes method
We could overload the function to have one function that takes just hours, and another with hours and mins, or have just one that takes hours and minutes, but make the minutes optional and default to zero

BTW. I don't think defining loads of constants for each timezone would end up taking any Flash as the compiler would just replace the constant with its value in the binary.

However maintaining the list of constants would be a problem, as I'd not be surprised if these things change from time to time because of geopolitical changes
or

csnol, 2017-10-19:
How about that ?
``` c++
// usage: localtime = TimeZone(UnixTime, 8, 45);   ->  UTC +8:45
time_t TimeZone(time_t t, int HTZ, int MTZ ) { return ( t + (TZ * SECS_PER_HOUR) + (MTZ * 60)); } 
```

staticmem, 2017-10-19:
What is the drama with using a Double or better still a Float depending on the processor?  The variable space required is pushed onto the stack and released after the function exits.  Is the real issue extra code space required to handle floats?  One could use Fixed point variables but I'm sure that would require more code space.

rogerclarkmelbourne, 2017-10-19:
I know @stevstrong  has a conceptual problem with using float or double for this, but unless someone is calling this function a lot in their program, I can't see how its going to have a serious impact

I would be interesting to see which results in the smaller code size i.e separate hours and mins vars etc or just one float (or double) value

But float (or double) will definitely be slower

abouillot, 2017-10-19:
I know in some systems the time zones are handled in multiple of quarters
of hours. This means 4 will represent a time difference of one hour, 32 for
8 hours and 7 would represent 1h45. I don't have any idea of the impact of
this in the code, but you remain in integers world.

stevstrong, 2017-10-19:
> How about that ?
> 
> ```
> // usage: localtime = TimeZone(UnixTime, 8, 45);   ->  UTC +8:45
> time_t TimeZone(time_t t, int HTZ, int MTZ ) { return ( t + (HTZ * SECS_PER_HOUR) + (MTZ * 60)); } 
> ```
It looks fine for me.

csnol, 2017-10-19:
I've open a PR for this issue. 
Integer TZ use:  TimeZone(UnixTime, 8);  or TimeZone(UnixTime, 8, 0);
Non-integer TZ use: TimeZone(UnixTime, 8,45);

staticmem, 2017-10-19:
I never thought that my issue would create so much interest!  I appreciate all the input to address this particular issue.

csnol, 2017-11-23:
You're Welcome.  Have fun in Cortex Mx world.  
 if not problems, Please close this issue . @staticmem  @rogerclarkmelbourne 
Thanks again for bug checked.

closed 2019-01-28 by rogerclarkmelbourne