rogerclarkmelbourne/Arduino_STM32 #370: ST-LINK_USB_V2_Driver is old

nanbuwks, 2017-11-04:
\tools\win\stlink\ST-LINK_USB_V2_Driver contents,
On this repository version seems very old version.
It not does not correspond Windows10.

It seems STSW-LINK009.
http://www.st.com/ja/development-tools/stsw-link009.html
It is more new version , available Windows10.

rogerclarkmelbourne, 2018-02-12:
I'm not sure if the license on this allows it to be put on github

See

http://www.st.com/content/ccc/resource/legal/legal_agreement/license_agreement/group0/e5/31/22/e2/88/a4/45/d8/DM00217720/files/DM00217720.pdf/jcr:content/translations/en.DM00217720.pdf

The license seems to allow redistribution as part of a  "Product" and defines "Product" as 

> Product: means Your and Your’s end-users’ product or system

So if this repo is a "Product" it looks like it would be OK, however I'm not sure if this community development qualifies as a Product.

perhaps @fpistm would know

BTW.
I have no record of there being any license for the old driver, at least not packaged with the files when I downloaded them 3 years ago.

fpistm, 2018-02-13:
@rogerclarkmelbourne 
you could redistribute it, it is also done with [Arduino_Tools](https://github.com/stm32duino/Arduino_Tools)

rogerclarkmelbourne, 2018-02-13:
@fpistm 

Thanks Frederic

I'll update it

rogerclarkmelbourne, 2019-01-30:
Fixed in  https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/8dd0bec98e18d84abe984caa3d061d0dacb5b31d

commit 8dd0bec9 (2019-01-30): Update STLink and Flashloader tools to fix #370
commit c1aba241 (2019-02-03): Update STLink and Flashloader tools to fix #370
closed 2019-01-30 by rogerclarkmelbourne