rogerclarkmelbourne/Arduino_STM32 #372: Use 115200 baud for serial upload

arpruss, 2017-11-06:
I would suggest using 115200 baud by default for the serial upload option, instead of the 230400 currently used by the serial_upload scripts, because 115200 the highest speed officially tested to work with the ROM bootloader according to [the STM docs](http://www.st.com/content/ccc/resource/technical/document/application_note/51/5f/03/1e/bd/9b/45/be/CD00264342.pdf/files/CD00264342.pdf/jcr:content/translations/en.CD00264342.pdf). And indeed on my stm32f103c, the default 230400 did not work. It's only a small increase in upload time for a typical sketch, and a decrease in headaches for users in trying to track down which script to modify.

rogerclarkmelbourne, 2017-11-06:
I could change this, but STM's own uploader program has speeds above 115200.

It has 256000 as an option

I think the problem is that different USB to Serial adpator drivers support different sets of baud rates.

They all support the standard rates e.g 115200, but above that, some support 256000 where as some support 230400

I think I used 230400 because my USB to serial adaptor supported that speed and not 256000
(I think FTDI supports 256000)

Did you try 256000 ?

arpruss, 2017-11-07:
I am using a Mega clone, which has a ch340. The web says that a ch340 that's part of an Arduino only goes up to 115,200 (maybe something to do with not having its own clock). 

In any case, it might be better to do 115,200, because that's probably the highest speed that is consistently supported by the various adapters.

stevstrong, 2017-11-07:
I am with @arpruss this time.
Standard is better when lots of people use the tool with different serial adapters.
So +1 for 115200 from me.

arpruss, 2017-11-07:
Yes: people who have setups that go beyond the standard can edit their scripts to optimize upload speed, but it would be nice if things worked out of the box for everybody.

rogerclarkmelbourne, 2017-11-08:
Should be fixed by 

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/c5ed88921ef7228bdd8decf9711e98536c3bd6a1

Note. I updated the files on Windows, so I don't know if the permissions get trashed for OSX and Linux.

If they do, I'll need to clone the repo onto a linux box etc and re-instate the permissions, but thats a pain, so lets hope that the Windows github client has not changed the permissions

closed 2017-11-13 by rogerclarkmelbourne