rogerclarkmelbourne/Arduino_STM32 #374: Could not find boards.txt in...

SalvorinFex, 2017-11-13:
Howdy, I keep getting this error every time I load either a Board under Boards Manager or the Library Manager:

Could not find boards.txt in C:\Users\USER_NAME\Documents\Arduino\hardware\Arduino_STM32\examples. Is it pre-1.5?

Is this standard or something that can be fixed?

rogerclarkmelbourne, 2017-11-13:
Who's board manager package you using, as I don't have one in my repo ?

rogerclarkmelbourne, 2017-11-13:
What version of the IDE are you using ?

rogerclarkmelbourne, 2017-11-13:
Try downloading the repo again now.  There was a screw up with a PR for an example, and I'd not noticed that the author of the PR has put the example in the wrong place in the heirarchy

SalvorinFex, 2017-11-13:
Okay, that must have been it, the message isn't coming up now.

Thanks Roger, on top of things as always.

closed 2017-11-13 by SalvorinFex

Bigoldsfofty, 2019-09-18:
I get this error on a fresh install of IDE 1.8,10 when it just did a library update.
On Windows 10

Could not find boards.txt in :\Users\USER_NAME\Documents\Arduino\hardware\arduino\bootloaders. Is it pre-1.5?
Could not find boards.txt in C:\Users\USER_NAME\Documents\Arduino\hardware\arduino\cores. Is it pre-1.5?
Could not find boards.txt in C:\Users\USER_NAME\Documents\Arduino\hardware\arduino\variants. Is it pre-1.5?
Could not find boards.txt in C:\Users\USER_NAME\Documents\Arduino\hardware\arduino\bootloaders. Is it pre-1.5?
Could not find boards.txt in C:\Users\USER_NAME\Documents\Arduino\hardware\arduino\cores. Is it pre-1.5?
Could not find boards.txt in C:\Users\USER_NAME\Documents\Arduino\hardware\arduino\variants. Is it pre-1.5?

BTW I got an error from this site about Internet Explorer.  The reason I still us it is because it is faster than "The Edge" and Firefox on this machine.  Quite frankly I don't see why on should have to install 5 different browsers to keep a few web-developer happy

stevstrong, 2019-09-18:
You have to follow the [instruction how to install the core](https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation).

Bigoldsfofty, 2019-09-19:
The problem persists.  Why is it looking for boards.txt in those locations.
It is correct there is no such file.  Where should it be and what should be in it in this case?
There is a file by that name  
\Documents\Arduino\hardware\Arduino\avr

Where should it really be?

stevstrong, 2019-09-19:
If you would have followed the instructions, Arduino IDE should find it without any problem.
Alternative source for installation procedure: https://web.archive.org/web/20190417082408/http://wiki.stm32duino.com/index.php?title=Installation

TDHofstetter, 2022-11-30:
We need a generic fix, though, instead of just "FTFI". There are other boards that also involve the same error, even when they're installed according to instructions... and some that cannot be installed using those instructions yet they are perfectly viable boards and are fairly heavily used throughout ArduinoWorld. My Sanguino hardware installation, for example, gives me exactly the same error. Oh, it works perfectly well, of course... it just gives me that error at IDE startup.

So... somebody needs to fix something with the IDE. I'm not sure exactly what they need to fix, but it needs fixing. Also... we need a better description of the internals of the problem, and a better resolution to it, than "FTFI".

board707, 2022-12-03:
> My Sanguino hardware installation, for example, gives me exactly the same error.

So the problem is most likely in your computer, and not in this package. Why do you require a fix from the authors of this software?
I have supports of many families of boards in my Arduino IDE and all the boards files are in their places.  Arduino does not give me an error message.... 
I think you should look the solution at your side first.

TDHofstetter, 2022-12-03:
> So the problem is most likely in your computer, and not in this package. Why do you require a fix from the authors of this software? I have supports of many families of boards in my Arduino IDE and all the boards.txt files are in their places. Arduino does not give me an error message.... I think you should look the solution at your side first.

I think it's appropriate for me to "require" a fix here because I'm certainly not the only person who is experiencing this problem. I downloaded the Sanguino hardware profile from a standard trusted source, which means that a whole lot of other people also did the same thing. I installed it by standard methods, which means that a whole lot of other people did the same thing. It fails on my machine, which means that it failed on a whole lot of other peoples' machines.

The whole point of programming is... doing something once so that countless other people don't have to do it countless times.

You should perhaps also know that I have successfully built custom hardware profiles for unusual microcontrollers on custom, unique boards... and those profiles have NOT generated any warnings or errors, and they work as expected, so clearly nothing is wrong with my machine. The Arduino IDE simply doesn't tolerate the world-standard Sanguino hardware profile.

...and that's wrong.

...and so it would be good to at the very least get some sort of explanation as to the criteria which might make the Arduino IDE reject a seemingly perfectly valid hardware profile and pop that error message... especially in light of the fact that (at least as far as I have been able to test, which is <100%) the Sanguino profile in question is functional despite the error message.

... which is ALSO wrong.

When something generates an error message, it should be connected to something that fails to function, should it not? This doesn't seem to fail to function except inasmuch as it generates an apparently inappropriate error message.

Do you actually not concur?

Oh, I might add that I can find two copies of boards.txt, and two references to it/them, within hardware/Sanguino-master. The error message says that it cannot do so. Hmmm, don't you think?

board707, 2022-12-03:
> Oh, I might add that I can find two copies of boards.txt, and two references to it/them, within hardware/Sanguino-master. The error message says that it cannot do so. Hmmm, don't you think?

I think that you mixed up something, because you are writing here, although this package has nothing to do with Sanguino boards.
It is possible that when you installed the package, you also mixed up something - so it does not work for you

JiriBilek, 2022-12-03:
Sorry if I am missing something.
But it seems to me that Sanguino is a totally different repository and a totally different hardware. So it would be much more efficient to address your problems there.
This is a repository maintained by volunteers in their free time, so I doubt there would be someone who'd like to study another repo and fix it.
Happy coding.

TDHofstetter, 2022-12-03:
Guys... geez. We aren't talking about a firmware reposititory here, we're talking about an failure of the Arduino IDE, which blows up when something isn't the way it wants things to be, but can't be any more informative than it is.

Are we not all on the same page here?