rogerclarkmelbourne/Arduino_STM32 #378: SPI setBitOrder is missing

rogerclarkmelbourne, 2017-11-14:
SPI setBitOrder is missing

I'll investigate how to implement it

rogerclarkmelbourne, 2017-11-14:
Umm

Our implementation seems the same as the Arduino SAM core

But I get 

`D:\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:98:29: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'BitOrder' [-fpermissive]`

when compiling for the BluePill

SAM has BitOrder defined as

```
enum BitOrder {
	LSBFIRST = 0,
	MSBFIRST = 1
};
```

https://github.com/arduino/ArduinoCore-sam/blob/master/cores/arduino/wiring_constants.h#L46-L49

Which is the same as in LibMaple

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/cores/maple/wirish_constants.h#L8-L11

One way to fix this would be to have a version of setBitOrder that accepted a uint8_t argument

stevstrong, 2017-11-15:
I think you have to include 
`|| define (ARDUINO_ARCH_STM32)`
to line https://github.com/PaulStoffregen/RadioHead/blob/master/RHHardwareSPI.cpp#L106, then it should be fine.

rogerclarkmelbourne, 2017-11-15:
Thanks Steve, I'll try that

I tired a few things, but there seemed to be more and more compile errors as I went though the code fixing things one at a time.

It only just crossed my mind that the Teensy is not so different from STM32 so looking at their port may be the best place to start

victorpv, 2017-11-30:
What if we just add the BitOrder enum?
```
enum BitOrder {
	LSBFIRST = 0,
	MSBFIRST = 1
};
```

rogerclarkmelbourne, 2017-11-30:
Yes.
I noticed that as well.

I think the problem is something else in RadioHead, and after looking at the RadioHead code, which is packed full of ifdefs and has different files for different MCU's I don't think its worth the hassle of trying to figure out what it doesnt like with LibMaple

So I"ll close this issue

closed 2017-11-30 by rogerclarkmelbourne