rogerclarkmelbourne/Arduino_STM32 #382: Malfunctioned I2C bus will halt program

DeqingSun, 2017-11-17:
i2c_start_condition and i2c_stop_condition in i2c.h uses while loop. 

If I2C bus has hardware failure, it will never exit loop. Tie SCL to GND will cause dead loop in i2c_start_condition.  Tie SDA to GND will cause dead loop in i2c_stop_condition.

rogerclarkmelbourne, 2017-11-18:
What happens on an AVR Arduino if you repeat your test?

What would you recommend to fix this (assuming that the AVR Arduino does not hang in a loop) ?

lacklustrlabs, 2017-11-29:
Pulling down SCL and/or SDA with 1KΩ to ground, will hang `Wire.endTransmission()` on Arduino Pro Mini (Arduino 1.8.2)

I tested for 10+ minutes on each pin.

rogerclarkmelbourne, 2017-11-29:
Yes.

There is no timeout for incorrectly wired I2C devices

rogerclarkmelbourne, 2018-02-12:
@lacklustrlabs 

Just to confirm... I think you are saying the official AVR core has the same issue.

???

If so I can close this

dazhbog, 2018-04-10:
This is an expected behaviour on most arduinos. I2C hangs, busy device, device missing, etc. will result in a while loop.

I suggest you implement a timeout yourself or use software I2C.

rogerclarkmelbourne, 2018-05-23:
Sounds like this is not specifically an issue with this implementation of the Arduino API, if it also applies to AVR

closed 2018-05-23 by rogerclarkmelbourne