rogerclarkmelbourne/Arduino_STM32 #387: Need to update BMP upload settings in platform.txt

rogerclarkmelbourne, 2017-11-22:
I Need to update BMP upload settings in platform.txt using latest version posted by @RickKimball

stevstrong, 2017-11-22:
Please don't forget to change it for F4, too, if needed.

rogerclarkmelbourne, 2017-11-22:
OK. I'll update the F4 at the same time

For my reference the post from @RickKimball that contained the code was

http://www.stm32duino.com/viewtopic.php?f=37&t=2802&p=36891#p36891

stevstrong, 2019-09-16:
Roger, do you plan to do something with this?

rogerclarkmelbourne, 2019-09-16:
Looks like its documented here

https://web.archive.org/web/20190316162124/http://www.stm32duino.com/viewtopic.php?f=37&t=2802

rogerclarkmelbourne, 2019-09-16:
It would need to be tested by someone.  I will need to find a board with the BMP loaded.

Perhaps someone with a BMP can confirm this upload pattern is OK

rogerclarkmelbourne, 2020-05-18:
Closing this, as it does not appear to be a problem, as no one has complained about this since 2017

closed 2020-05-18 by rogerclarkmelbourne