rogerclarkmelbourne/Arduino_STM32 #393: Serial problem in PlatformIO

denisn73, 2017-11-29:
I'm try build genericSTM32F103Cx variant in PlatformIO and get errors:
---
- "src\main.cpp:5:18: error: no matching function for call to 'HardwareSerial::begin()'"
- "src\main.cpp:8:5: error: 'Serial3' was not declared in this scope"

main.cpp:
---
#include <Arduino.h>
void setup() {
    // put your setup code here, to run once:
    Serial.begin();
    Serial1.begin(115200);
    Serial2.begin(115200);
    Serial3.begin(115200);
}
void loop() {
    // put your main code here, to run repeatedly:
}

platformio.ini:
---
[env:genericSTM32F103CB]
platform = ststm32
board = genericSTM32F103CB
framework = arduino

...\.platformio\packages\framework-arduinoststm32.package.json:
---
{
    "description":"Arduino Wiring-based Framework (ST STM32 Core)",
    "name":"framework-arduinoststm32",
    "url":"https://github.com/rogerclarkmelbourne/Arduino_STM32",
    "version":"2.0.170723"
}

In ArduinoIDE all work fine.
===

stevstrong, 2017-11-29:
This core does not officially support PlatformIO.
Please use the forum for such questions.
And first check some forum posts before submitting any issue.
http://www.stm32duino.com/viewtopic.php?f=41&t=2691&p=36874&hilit=platformio#p36874

rogerclarkmelbourne, 2017-11-29:
I suggest you contact the PlatformIO team, as support in this repo is only for the Arduino IDE.
Not any third party IDE or dev system

closed 2017-11-29 by rogerclarkmelbourne