rogerclarkmelbourne/Arduino_STM32 #395: Can not find functions in RTClock library

SouperMan3, 2017-12-01:
In the RTClock library utilities  (rtc_util.c)  three start functions are listed:
rcc_start_lse(), rcc_start_lseI(), and rcc_start_hse(,,  I try to compile but these functions are not found by the compiler.  I looked in rcc.c  and did not see them   Please check this out & let me know where they are .
Thanks

lacklustrlabs, 2017-12-02:
rcc_start_lse(), rcc_start_lsi() and rcc_start_hse() is defined as static inline functions in these files:
```
STM32F1/system/libmaple/stm32f1/include/series/rcc.h
STM32F4/cores/maple/libmaple/rccF4.h
```
So the answer depends on what platform you are using.

lacklustrlabs, 2017-12-02:
rcc_start_lse(), rcc_start_lsi() and rcc_start_hse() are defined as static inline functions in these files:
```
STM32F1/system/libmaple/stm32f1/include/series/rcc.h
STM32F4/cores/maple/libmaple/rccF4.h
```
So the answer depends on what platform you are using.

rogerclarkmelbourne, 2017-12-03:
This does not seem to be a issue in the core, so I'm going to close it.

Re-open it if the core or a built in library does not compile

closed 2017-12-03 by rogerclarkmelbourne