rogerclarkmelbourne/Arduino_STM32 #396: HardWire.h: No such file or directory

tutankhamen, 2017-12-02:
Trying to use adafruit SSD1306 library for STM32F1, but getting this error. It references to HardWire.h file which doesn't exist.

rogerclarkmelbourne, 2017-12-02:
Wire now uses HardWire (and has for several months)

Whatever library you are using is out of date, and should be changed so that Wire is used instead of HardWire

tutankhamen, 2017-12-03:
Thank you for quick reply! This is what I've got today here: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/Adafruit_SSD1306/Adafruit_SSD1306_STM32.cpp

rogerclarkmelbourne, 2017-12-03:
OK

It didnt get updated when the to Wire was made

Can you try changing it, as I cant look at it until later today and I don't even have that display, All I could do is make sure it compiles

tutankhamen, 2017-12-03:
Yes, sure, what you want me to update?

tutankhamen, 2017-12-03:
I've replaced HardWire.h with Wire.h and it looks like it can be compiled now.

rogerclarkmelbourne, 2017-12-03:
Cool

Let me know if it works with those changes

I suspect that the declaration of HWire(2) can also be replaced unless you actually want to use I2C device 2

I'd try removing HWire(2); and do a global search and replace on HWire and replace by Wire

tutankhamen, 2017-12-03:
cool, thank you. It's compiling, but still not working :(  As I understand, first I2C on blue pill board is pins B8/B9, am I right?  Thank you for your support!

rogerclarkmelbourne, 2017-12-03:
See

http://wiki.stm32duino.com/images/a/ae/Bluepillpinout.gif

AFIK I2C is on PB6 and PB7

The image in the wiki is confusing as it lists both the default and "alternate" pins for I2C and SPI without saying which set of pins is the default

Look in the Maple mini pin list here

http://docs.leaflabs.com/static.leaflabs.com/pub/leaflabs/maple-docs/latest/hardware/maple-mini.html

It lists the pins in both its product specific Dx pin number and the STM official Pxy notation

tutankhamen, 2017-12-03:
Got it, but still nothing, tried both on 0x3D and 0x3C (but as i understand, 128x64 uses address 0x3D), use first i2c ( pb6 / pb7)

HardWire HWIRE(1,I2C_FAST_MODE); // I2c1

will try to investigate more...

Thank you!

rogerclarkmelbourne, 2017-12-03:
Try a normal I2C scanner, the one on the Arduino.cc website normally works

Make sure you have pullups to 3.3V and they have to be lower value than the normal 4.7k used on 5V

I think I normally use around 2.7k or whatever I have handy

rogerclarkmelbourne, 2017-12-03:
I've pushed a change to address the compile problems, but I don't have this display so I can't test it

lacklustrlabs, 2017-12-03:
I just tested both the ssd1306_128x32_i2c.ino and ssd1306_128x32_i2c_STM2.ino examples from the current master and an up-to-date Adafruit_SSD1306 library.

Maple mini, 4K7 Ohm pull-ups at PB_7 (SDA) and PB_6 (SCL).  
Both examples worked w/o having to change anything.

The OLED I tested doesn't have the reset pin, so OLED_RESET pin was left dangling.

Hm.. why does those I2C examples include SPI.h?

tutankhamen, 2017-12-03:
Yay! I replaced 4.7kOhm resistors with 2.2kOhm and everything is start working! Thank you!

rogerclarkmelbourne, 2017-12-03:
@tutankhamen 
Did you try it with the latest code I posted ?

@lacklustrlabs 
I agree this library needs to be tidied up.

I didn't look that closely at the code when I added/merged  it, I now see it does seem to have some redundant parts.

tutankhamen, 2017-12-03:
Nope, I've used the old one with my modifications. Will try the new one now.

tutankhamen, 2017-12-03:
Just tested with the new version - works just perfect without modifications! Thank you!

rogerclarkmelbourne, 2017-12-03:
Excellent.

Thanks for testing

rogerclarkmelbourne, 2017-12-10:
FYI.

Things have changed again, as there was an issue with the name HardWire for the class, so this has now been changed to TwoWire to match the internal Arduino class name for Wire

closed 2017-12-10 by rogerclarkmelbourne