rogerclarkmelbourne/Arduino_STM32 #398: Serial1/2/3 with HC-05 bluetooth module with stm32f103

tutankhamen, 2017-12-05:
Trying to use my HC-05 bt module with stm32f103 to exchange data with Android device , but have some weird behavior with incoming data. I made a simple test which is  sending 10 bytes string "0123456789" from android to stm32, but I'm receiving only random 1-5 bytes. Code on stm32 side:

`long received = 0;`
`const unsigned long buffer_size = 128;`
`uint8_t buffer[buffer_size];`

`void setup() {`
`    Serial1.begin(9600);`
`}`

`[...]`
`void loop() {`
`    [...]`
`    if(Serial1.available()){`
`        received += Serial1.readBytes(buffer + received, bufferSize-received);`
`    }    `
`}`

So, if will send "0123456789" from android, let say, 3 times every 1 second, on stm32 i will receive 3 portions like this:

"0", "0567", "047"

or something like this, i.e. I always receive first byte and it's always correct, but it looks like the rest bytes could be randomly lost. If I'm sending data from stm32 to Android - everything works just fine. Also, I tried to use the same code with Arduino nano - everything works just perfect, i.e. I could transfer the data in both directions without any problem. Also tried to use Serial2 and Serial3, tried to change bod rate.

Android code looks like this:

`BluetoothSocket socket;`
`OutputStream outStream;`
`[...]`
`void init(...) {`
`    socket = ConnectToSTM32(...);`
`    [...]`
`    outStream = socket.getOutputStream();`
`}`
`void send(byte[] bytes) {`
`    outStream.write(bytes);`
`}`
`[...]`
`String strToSend = "0123456789";`
`send(strToSend.getBytes());`

Wiring is like this:

[HC-05]      [STM32]
  GND    ---   GND
  +5V     ---   VCC
   RXD    ---   PA9
   TXD    ---   PA10

stevstrong, 2017-12-05:
Please use the forum for such issues: www.stm32duino.com

tutankhamen, 2017-12-05:
sorry. do you want me to delete this ticket?

stevstrong, 2017-12-05:
Yes, please.

tutankhamen, 2017-12-05:
I can delete comments, but not the ticket :(

stevstrong, 2017-12-05:
It should be possible to close the issue.

rogerclarkmelbourne, 2017-12-05:
I'll close it

closed 2017-12-05 by rogerclarkmelbourne