rogerclarkmelbourne/Arduino_STM32 #402: error: 'constexpr' needed for in-class initialization of static data member -- while compiling STM32ADC example

tfry-git, 2017-12-08:
When including the header file Arduino_STM32/STM32F1/libraries/STM32ADC/src/STM32ADC.h - e.g. while trying to build one of the SingeConversionInterrupt example - I get the following error:

```
In file included from /bla/hardware/Arduino_STM32/STM32F1/libraries/STM32ADC/examples/SingleConversionInterrupt/SingleConversionInterrupt.ino:1:0:
/bla/hardware/Arduino_STM32/STM32F1/libraries/STM32ADC/src/STM32ADC.h:155:40: error: 'constexpr' needed for in-class initialization of static data member 'const float STM32ADC::_AverageSlope' of non-integral type [-fpermissive]
     static const float _AverageSlope = 4.3; // mV/oC   //4.0 to 4.6
```

Also a instance of the same error for the header line immediately after that. Compiler:
`arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors (Arduino build)) 4.8.3 20140228 (release) [ARM/embedded-4_8-branch revision 208322]`

See also http://www.stm32duino.com/viewtopic.php?f=14&t=2207 . Changing "const" to "constexpr" in this and the following line allows compilation to proceed.

lacklustrlabs, 2017-12-08:
Confirmed. Using `-std=c++98` makes the error go away, so it was probably a side effect to the `-std=gnu++11` transition.

g++ 7.2 and 6.3 reports the same error (gnu++11).

tfry-git, 2018-01-24:
Fixed by https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/407.

closed 2018-01-24 by tfry-git