rogerclarkmelbourne/Arduino_STM32 #403: util/atomic.h is missing

tfry-git, 2017-12-08:
The header file <util/atomic.h> is part of avr-libc (http://www.nongnu.org/avr-libc/user-manual/group__util__atomic.html) and thus may be assumed by arduino libraries. One example library using atomic.h is Mozzi.

Essentially it allows to block interrupts during certain code sections, and re-enabling them afterwards. While the need for such functionality may be more pronounced on 8-bit CPUs and their need for multi-step calculations, there are valid usecases for stm32 chips as well.

User "pico" has submitted a port of atomic.h at http://www.stm32duino.com/viewtopic.php?f=3&t=258&start=10 (see also the surrounding discussion in that thread), but it seems to have been forgotten.

I am using his code successfully, locally on an STM32F103C8T6. I can confirm that
- it compiles
- code using it works without obvious defects (the Mozzi library; and yes, I made sure to specifically test some functions that make use of "ATOMIC_BLOCK")
- I did not test, whether it actually protects from intervening interrupts (I have no reason to believe, otherwise).
- I did test to make sure that interrupts continue to work as expected outside the protected code (i.e. after ATOMIC_RESTORESTATE was called)

tfry-git, 2017-12-08:
Testcase for the patch. Works as expected (LED permanently on) on STM32F103C8T6 "blue pill".

```
#include <HardwareTimer.h>
#include <util/atomic.h>

HardwareTimer timer(1);

volatile uint32_t count;
volatile uint32_t dummy;

uint32_t countInterrupts () {
  count = 0;
  // This loop should take more than just a few microseconds, but not too long for us to wait
  for (uint32_t i = 0; i < (1 << 24); ++i) {
    dummy = i; // So compiler won't try to outsmart us
  }
  return count;
}

void timerInt () {
  ++count;
}

void setup() {
  pinMode(PC13, OUTPUT);

  timer.pause ();
  timer.setPeriod (10);
  timer.setChannel1Mode (TIMER_OUTPUT_COMPARE);
  timer.setCompare (TIMER_CH1, 1);
  timer.attachCompare1Interrupt(timerInt);
  timer.resume ();
}

void loop() {
  uint32_t a, b;
  ATOMIC_BLOCK(ATOMIC_RESTORESTATE) {
    a = countInterrupts ();
  }
  digitalWrite (PC13, a > 0);
  b = countInterrupts ();
  digitalWrite (PC13, b < 1000);
}
```

lacklustrlabs, 2017-12-08:
Slightly off topic:
I wonder if we could combine this with a construct similar to [std::lock_guard](http://en.cppreference.com/w/cpp/thread/lock_guard)?
For simplicity sake a lock_guard that operates on an un-named global mutex (Cli/Cei).

RickKimball, 2017-12-08:
> The header file <util/atomic.h> is part of avr-libc (http://www.nongnu.org/avr-libc/user-manual/group__util__atomic.html) and thus may be assumed by arduino libraries. One example library using atomic.h is Mozzi.

This is a completely bogus assumption.  This core doesn't use the avr libc and you shouldn't expect it.

tfry-git, 2017-12-08:
> This core doesn't use the avr libc and you shouldn't expect it.

Well, sorry, I wasn't being too precise. Libraries written for AVR may assume them, and having a source-compatible define will make it easier to port them. Plus, as mentioned, the _feature_ (irrespective of API details) may come in handy on any CPU.

RickKimball, 2017-12-08:
gcc has built-in atomic stuff. Wouldn't it be better to focus using that?

https://gcc.gnu.org/onlinedocs/gcc-4.8.3/gcc/_005f_005fatomic-Builtins.html

stevstrong, 2017-12-08:
The GCC atomic stuff does not cover the matter presented here, which I recognize can be useful under certain circumstances.
So I vote +1, also because does not remove or change anything, just adds some utility functions.

RickKimball, 2017-12-08:
When the original author of Mozzi ported to the Teensy 3.0/3.1 why didn't he create a util/atomic.h?  He took a different approach.  In addition, he lists a litany of changes and limitations:

http://sensorium.github.io/Mozzi/blog/2014/08/09/teensy-3-dot-0-slash-3-dot-1-test-release-version-2014-08-09-14-15/

Why should the stm32 port use your approach?

tfry-git, 2017-12-08:
> When the original author of Mozzi ported to the Teensy 3.0/3.1 why didn't he create a util/atomic.h?

Because he did. In the teensy core files:

https://github.com/PaulStoffregen/cores/blob/master/teensy3/util/atomic.h

RickKimball, 2017-12-08:
yes I just realized that.  I guess this is just me be generally against disabling interrupts when we have so many peripherals that allow you to run without disabling them.

lacklustrlabs, 2017-12-08:
@tfry-git  regarding that test case, I assume you mean 'LED permanently **off**'?

If we do merge this, can we add a non-C-style alternative as well? Something like:

```
template<bool restoreState=true>
struct InterruptLockGuard {
  inline InterruptLockGuard() __attribute__((always_inline)) {
    if (restoreState) {
      prevStatus = interruptsDisabled();
    }
    noInterrupts();
  }

  InterruptLockGuard(const InterruptLockGuard&) = delete;
  InterruptLockGuard& operator=(const InterruptLockGuard&) = delete;
  InterruptLockGuard(InterruptLockGuard&&) = delete;
  InterruptLockGuard& operator=(InterruptLockGuard&&) = delete;
    
  inline ~InterruptLockGuard() __attribute__((always_inline)) {
    if (restoreState) {
      if (!prevStatus){
        interrupts();
      }
    } else {
      interrupts();
    }
  }
  
  inline bool interruptsDisabled() const __attribute__((always_inline)) {
    uint32_t primask;
    asm volatile("mrs %0, primask\n" : "=r" (primask)::);
    return primask;
  }
  
  private:
  bool prevStatus;
};
```

It works like std::lock_guard but uses primask as mutex (and it only blocks interrupts)

```
void loop() {
  uint32_t a, b;
  {
    InterruptLockGuard<true> ilg;
    a = countInterrupts ();
  }
  digitalWrite (PC13, a > 0);
  b = countInterrupts ();
  digitalWrite (PC13, b < 1000);
}
```

tfry-git, 2017-12-09:
> regarding that test case, I assume you mean 'LED permanently off'

It's one of those boards, where the LED in on when PC13 is _LOW_. Both tests (a > 0, b < 1000) evaluate to false.

rogerclarkmelbourne, 2017-12-10:
Is someone going to submit a  PR for this change ?

tfry-git, 2017-12-11:
With https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/1cc4f791914c20e4c562c8778ba3b41e3a0063d7 this now works as (I) exptected (it).

Thanks!

closed 2017-12-11 by tfry-git