rogerclarkmelbourne/Arduino_STM32 #412: Timer interrupt flag issue introduced in commit dfb5b4=update timer.h

pascallanger, 2017-12-19:
Hi,
[This change](https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/dfb5bd4c30f42fa9cf2a0aef6b7d2a50a594fa97) is causing an issue. While it's totally fine in theory to clear the interrupt flag before enabling the interrupt, it has the downside of reading, modifying and writting the SR register (even with bit banding). If at the same time SR is also modified by a comparator on a different channel then this interrupt flag is lost. I agree this is a race condition that can happen only rarely but we've learned it the hard way within our [multiprotocol project](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module) using stm32duino as a base which has caused some planes to crash...
You need to only write to the SR register and not read it when you want to update it. The fix applied has been to use:
TIMER2_BASE->SR = 0x1E5F & ~TIMER_SR_CC1IF;
This is ok since the SR register is specified as type "rc_w0".
The definition of "rc_w0" is:
"read/clear (rc_w0) Software can read as well as clear this bit by writing 0. Writing ‘1’ has no effect on the bit value."
So the fix is to write 1 everywhere (except in the reserved area which are initialized to 0 at reset => 0x1E5F) and a 0 only at the exact flag location you want to clear.
Pascal

stevstrong, 2017-12-19:
I see your point.
The change I committed was tested when the timer is first stopped and then was the interrupt attached.
So if you first stop the timer, there is no issue, it cannot ignore/overwrite some bits during RMW (read-modify-write) process.
However, I also agree that your variant is much cleaner.

I think a working solution would be something like:
`dev->regs->SR = (~BIT(interrupt)) & ( (dev->type==TIMER_ADVANCED) ? 0x1EFF : 0x1E5F);`

pascallanger, 2017-12-19:
Looks good to me.

victorpv, 2018-04-12:
Did you guys test this, and if so did you submit a PR?

stevstrong, 2018-04-12:
Just pushed [#491](https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/491)

rogerclarkmelbourne, 2018-04-12:
Just merged #491

stevstrong, 2018-04-12:
Can be closed? Or OP wants to test?

rogerclarkmelbourne, 2018-04-12:
I'll close it and if the OP has a problem when he tests he can re-open

rogerclarkmelbourne, 2018-04-12:
@stevstrong 

I just retested and the Core does not compile with this change in #491 I will need to revert it :-(

stevstrong, 2018-04-12:
Ok, I'll check it again.

rogerclarkmelbourne, 2018-05-26:
Any news on this ?

stevstrong, 2018-05-26:
#497 fixes #491 and #412.
I suggest merging #497, then closing both #412 and #497.

rogerclarkmelbourne, 2018-05-26:
OK

I'll merge #497 and close the others

Thanks

rogerclarkmelbourne, 2018-05-27:
#497 has now been merged 

I did a general test and the core seemed to compile OK

closed 2018-05-27 by rogerclarkmelbourne