rogerclarkmelbourne/Arduino_STM32 #418: Error on First Compile

adityaalviori, 2017-12-25:
Dear All, 
I am fresh on STM32 and I had trouble when I compile on first time. I got error like that

In file included from C:\Users\Vio\Downloads\arduino-1.6.9\hardware\Arduino_STM32\STM32F1\cores\maple/Arduino.h:30:0,

                 from sketch\sketch_dec25a.ino.cpp:1:

C:\Users\Vio\Downloads\arduino-1.6.9\hardware\Arduino_STM32\STM32F1\cores\maple/wirish.h:41:20: fatal error: stdlib.h: No such file or directory

 #include <stdlib.h>

                    ^

compilation terminated.

exit status 1
Error compiling for board Generic STM32F103C series.

Thanks for your help.

lacklustrlabs, 2017-12-25:
I got no experience running this under Windows, but could you take a look and see if the stdlib.h file exists? It is installed by the boards manager (when you install Arduino Zero). Under linux and mac it can be found in the .arduino15 folder:
```
 $ find .arduino15/ -name stdlib.h
.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/machine/stdlib.h
.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/stdlib.h
.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/tr1/stdlib.h
```

rogerclarkmelbourne, 2018-01-08:
Sounds like an installation  problem.

If this issue does not get any traffic in the next 7 days I'll close it

stevstrong, 2018-01-24:
@adityaalviori 
It seems that you have put the Arduino_STM32 under the wrong directory.
It should be placed under `/user/Documents/Arduino/hardware` and not under the Arduino _installation_ directory, as your log shows.

rogerclarkmelbourne, 2018-02-12:
Closed due to lack of confirmation from @adityaalviori

wybiral, 2019-02-28:
I'm getting the same error on a fresh 1.8.8 Arduino IDE on 64bit Ubuntu Linux:
```
In file included from /home/davy/Downloads/arduino-1.8.8-linux64/arduino-1.8.8/hardware/Arduino_STM32-master/STM32F1/cores/maple/Arduino.h:30:0,
                 from sketch/sketch_feb28a.ino.cpp:1:
/home/davy/Downloads/arduino-1.8.8-linux64/arduino-1.8.8/hardware/Arduino_STM32-master/STM32F1/cores/maple/wirish.h:41:20: fatal error: stdlib.h: No such file or directory
 #include <stdlib.h>
                    ^
compilation terminated.
exit status 1
Error compiling for board Generic STM32F103C series.
```

It doesn't seem to matter which /hardware folder it's installed in.

rogerclarkmelbourne, 2019-02-28:
@wybiral 

Can you confirm which ARM compiler you have installed and how you installed it.

e.g. via the boards manager , by installing the Due etc

wybiral, 2019-03-01:
Hey @rogerclarkmelbourne 

0. I've started over with a fresh environment
1. Installed Arduino
2. Copied this repo to my `./hardware` folder
3. Ran `./tools/linux64/install.sh`

Now when I try to compile a sketch I get:
```
fork/exec /bin/arm-none-eabi-g++: no such file or directory
Error compiling for board Generic STM32F103C series.
```

I realize it's off topic for this issue but what's the correct way to configure the path to my arm-none-eabi directory?

Whatever I was doing the first time was probably what caused the original error (not sure how that relates to the OP error for this issue though).

rogerclarkmelbourne, 2019-03-01:
That installer just setups up linux specific stuff associated with the udev rules needed for the serial terminal etc

See the readme / wiki about installation. 

You need to install one of the Arduino boards that uses the ARM compiler, otherwise you won't have the ARM compiler (unless you installed one for something else, and even then the paths would most likely be wrong, and it may not be the correct version of compiler)

wybiral, 2019-03-01:
Thank you! I don't want to be a bother in a now unrelated issue but I really want to use your code (especially with the Adafruit_ILI9341_STM library) in an upcoming video project.

It compiles the sketch, which is the furthest I've gotten so far, but doesn't want to upload.
```
Sketch uses 14920 bytes (22%) of program storage space. Maximum is 65536 bytes.
Global variables use 3088 bytes (15%) of dynamic memory, leaving 17392 bytes for local variables. Maximum is 20480 bytes.
2019-02-28T20:29:27 INFO src/stlink-common.c: Loading device parameters....
2019-02-28T20:29:27 INFO src/stlink-common.c: Device connected is: F1 Medium-density device, id 0x20036410
2019-02-28T20:29:27 INFO src/stlink-common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
open() == -1
2019-02-28T20:29:27 ERROR src/stlink-common.c: map_file() == -1
stlink_fwrite_flash() == -1
```
Could this be some kind of permission error manifesting from the open() call?

For some reason the .json method from stm32duino works on my machine out of the box but I haven't been able to figure out how their configuration differs from this repo (either to use this TFT library on that core or get this core running).

rogerclarkmelbourne, 2019-03-01:
I don't use STLink on Linux. So I can't assist with this

(STLink on windows is different as it uses ST's own official CLI EXE. 

On Linux this core uses the open source Texane STlink system, but its possible its incompatible with your version of ST-Link.

ST may now have their own official STLink CLI for Linux which you could use to manually upload

stevstrong, 2019-09-16:
As it was long time inactive, can this issue be close?

rogerclarkmelbourne, 2020-02-06:
Closing because of no activity

closed 2020-02-06 by rogerclarkmelbourne