rogerclarkmelbourne/Arduino_STM32 #422: volatile qualifier

aster94, 2018-01-08:
hello to all!

I was trying to use this library: https://github.com/bolderflight/uNavAHRS
which needs this other lib: https://github.com/bolderflight/Eigen

but a lot of errors from the maple core appear:

`In file included from /home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/cores/maple/boards.h:37:0,
                 from /home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/cores/maple/wirish.h:54,
                 from /home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/cores/maple/Arduino.h:30,
                 from /tmp/arduino_build_821086/sketch/uNavAHRS_MPU9250.ino.cpp:1:
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'const std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/bits/locale_facets.h: In member function 'std::num_get<_CharT, _InIter>::iter_type std::num_get<_CharT, _InIter>::get(std::num_get<_CharT, _InIter>::iter_type, std::num_get<_CharT, _InIter>::iter_type, std::ios_base&, std::ios_base::iostate&, bool&) const':
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: expected primary-expression before 'volatile'
[... 195 lines cut ...]
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: ISO C++ forbids declaration of 'type name' with no type [-fpermissive]
 #define __io volatile
              ^
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: ISO C++ forbids declaration of 'type name' with no type [-fpermissive]
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: expected primary-expression before 'volatile'
/home/vincenzo/.arduino15/packages/stm32duino/hardware/STM32F1/2018.1.3/system/libmaple/include/libmaple/libmaple_types.h:55:14: error: expected ')' before 'volatile'
In file included from /home/vincenzo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/functional:53:0,
                 from /media/vincenzo/HardDisk/Storage/Programmi/Arduino/libraries/Eigen/Eigen/Core:262,
                 from /media/vincenzo/HardDisk/Storage/Programmi/Arduino/libraries/Eigen/Eigen.h:61,
                 from /media/vincenzo/HardDisk/Storage/Programmi/Arduino/libraries/IMU-uNavAHRS/uNavAHRS.h:41,
                 from /tmp/arduino_modified_sketch_812165/uNavAHRS_MPU9250.ino:24:
/home/vincenzo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/typeinfo:39:37: error: expected ')' before end of line
 #pragma GCC visibility push(default)
                                     ^
/home/vincenzo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/typeinfo:39:37: error: expected primary-expression before end of line
/home/vincenzo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/typeinfo:39:37: error: expected ';' before end of line
/home/vincenzo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/typeinfo:39:37: error: expected '}' before end of line
/home/vincenzo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/typeinfo: At global scope:
/home/vincenzo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/typeinfo:39:37: error: expected '}' before end of line
/home/vincenzo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/typeinfo:39:37: error: expected declaration before end of line
exit status 1`

really it seems that the error is only one ` 'volatile' qualifiers cannot be applied to 'std::ios_base&`

any idea?

stevstrong, 2018-01-08:
First, this is not the right place to discuss error messages given by the compiler, please the forum: www.stm32duino.com

Second, this github place is reserved for the Arduino_STM32 core inherited from Leaflabs and supported by Roger.
However, it seems that you are using another core, [the official ST core](https://github.com/stm32duino/Arduino_Core_STM32) installed over the board manager.

So please place your question either in forum or on the right Github place.

aster94, 2018-01-08:
steve where exactly do you see that i am using the official st core?
there is wrote libmaple quite everywhere, second, i never installed the official core
so i think this is the right place to discuss this problem since it seems to be an error inside the libmaple, but if you say that i will write in the forum

stevstrong, 2018-01-08:
@aster94, you wrote:

> [i am using the last version of stm32duino _**from the board manager**_ and a bluepill]

The board manager includes the official ST core. Or is this information incorrect?

stevstrong, 2018-01-08:
The libmaple core you have to install it manually, download zip end extract it to /Arduino/hardware directory.

aster94, 2018-01-08:
> Or is this information incorrect?
correct but also this repo is installable with the board manager
http://dan.drown.org/stm32duino/package_STM32duino_index.json

stevstrong, 2018-01-08:
THIS IS the ST official CORE!!! Not the libmaple core!!!

rogerclarkmelbourne, 2018-01-08:
Dan's  board manager files, do install this repo, but he does not actively maintain this, and I don't have time to maintain a boards manager package for this, because the repo gets updated very often

Looking in the JSON files, the version seems to date to November 2016

Please install the repo manually as described in the link that Steve posted.

Also, uninstall the board manager package version

closed 2018-01-08 by rogerclarkmelbourne

aster94, 2018-01-09:
dan mantain it constantly, its updated many many times a months
[image]
last one is only 6 days old

rogerclarkmelbourne, 2018-01-09:
OK

I was not aware that Dan updates it that often.
I know he has some scripts to do it, but it still takes some effort.

Nevertheless, as some issues may occur because of the boards manager and file location etc etc.

I don't process issues on here unless they are a direct installation of the repo

shujaatak, 2018-01-18:
@aster94 did you solve the issue? I am facing the same issue and I'm wondering which steps should I follow to resolve those errors :(

shujaatak, 2018-01-18:
@rogerclarkmelbourne I have followed the exact same steps mentioned in https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation, BUT I still face the same error. Can you please recommend any solution?

aster94, 2018-01-18:
No i haven t even tried.
You should be able to solve this by correctly defining "__io volatile" inside libmaple_types.h
Do a pull request when you solve it!

shujaatak, 2018-01-18:
@aster94 I will update you if I find a solution.

shujaatak, 2018-01-18:
@stevstrong In the forum, in which section should I post this question?

aster94, 2018-01-18:
Steve, i don t understand your refuse to accept that this is a proven issue

shujaatak, 2018-01-18:
@stevstrong @rogerclarkmelbourne I have posted in the forum here: http://www.stm32duino.com/viewtopic.php?f=2&t=3108
I would highly appreciate if you check this out. 
Thanks!

stevstrong, 2018-01-18:
There is no **proven** issue yet which is related to this core.

@aster94 , you used a non-standard core installation (traces of PlatformIO are detected from your log), from there you can get unpredictable issues.
Those issues do not belong here, are not supported.
As Roger already stated, please delete `/user/documents/.arduino15` folder.
Please use the installation procedure from the wiki, and post in the forum if you have any problems.

@shujaatak , I will reply to your post in the forum.

shujaatak, 2018-01-18:
@stevstrong Would you please kind enough to test the code on your end? You may already know that even if one don't have an actual board, one can compile the sketch. This way you may be able to verify if the problem really exists.
All you need is to copy these three libraries to your Ardiuno\libraries:
https://github.com/bolderflight/uNavAHRS
https://github.com/bolderflight/Eigen
https://github.com/bolderflight/MPU9250

stevstrong, 2018-01-18:
See my comments in the forum.
Please post any further comments in the forum.

aster94, 2018-01-18:
Steve i m not using platformIO, i use the arduino ide
Anyway all these library are not needed. Just install eigen and run the example if you really don t trust us that this is a core problem

stevstrong, 2018-01-18:
See my comments [in the forum](http://stm32duino.com/viewtopic.php?f=2&t=3108&p=40000#p40000).
### Please post any further comments in the forum.

shujaatak, 2018-01-18:
@aster94 IMHO it's problem with the core!

shujaatak, 2018-01-18:
@stevstrong @rogerclarkmelbourne  @aster94 
The author of the uNavAHRS [says](https://github.com/bolderflight/uNavAHRS/issues/1#issuecomment-358699700): **I'm using GCC 5.4 with an ARM (Teensy 3.6) target.** 
I also have used GCC 5.4 to compile the Eigen library but it does not compile and show those same errors, from this I guess there is problem in Arduino_STM32.

aster94, 2018-01-18:
@shujaatak,even if i agree with you do as stevestrong said
They (he, roger, and many other) do this for free, they don t see a single euro from here so please follow their suggestions

shujaatak, 2018-01-18:
@aster94  I wholeheartedly appreciate what they are doing. I don't know if you are following that forum or not but I have already posted on forum since stevestrong told me. 
As I'm losing hope so I think soon I will quit working on this library.

stevstrong, 2018-01-18:
See solution [in forum thread](http://www.stm32duino.com/viewtopic.php?f=2&t=3108&p=40044#p40044).