rogerclarkmelbourne/Arduino_STM32 #426: Using internal oscillator instead of external

me21, 2018-01-16:
Please add an option to use internal oscillator instead of external for a board. Or, an ability to freely configure board clocks.

Right now, every variant defines `init` function which calls `setup_clocks` and it cannot be overridden.

lacklustrlabs, 2018-01-24:
I don't know about the external clock, but its easy to _set_ a custom clock speed. 
(If it actually works is another problem..)

To set a custom clock speed for a maple mini you can create a STM32F1/boards.local.txt file:
```
mapleMini.menu.cpu_speed.speed_custom=My custom CPU clock
mapleMini.menu.cpu_speed.speed_custom.build.f_cpu=48000000L
``` 

Restart arduino and the new option should appear.

@RickKimball has some useful settings [here](https://gist.github.com/RickKimball/71a388efd58ac0d3a0e1a272ecf4e095/)

rogerclarkmelbourne, 2018-01-25:
I think he means to not require the external crystal at all.

So its not just a matter of changing that setting in boards.txt, there needs to be changes to the clock source for the PLL etc

me21, 2018-01-25:
@rogerclarkmelbourne

You've understood me correctly. As for PR, I have more pressing issues at the moment. But perhaps I will return to this in the future.

rogerclarkmelbourne, 2018-01-25:
No worries.
I agree its a good idea to be able to specify the clock input in the variant, but everyone has more pressing things to do.

I'll close this issue and you can re-open it when/if you submit a PR or need clarification

closed 2018-01-25 by rogerclarkmelbourne

RickKimball, 2018-01-25:
Doesn't the Nucleo_F103rb configuration already provide the feature?

me21, 2018-01-25:
It does; as far as I understand, it is controlled by certain preprocessor define. It would be great to make something generic for all configurations.