rogerclarkmelbourne/Arduino_STM32 #432: Serial.begin(xxxx); tries to send on SPI or SPI2

artbody, 2018-01-26:
Very strange behaviour with an stm32f103rb nucleo 64
after installing this Arduino_STM32
no example from this branch works 
all examples with Serial.begin(); are trying to send over the SPI 
a
 `Serial.println("Simple test");`
prints nothing over the serial interface (USB) to serial monitor (arduino IDE)

rogerclarkmelbourne, 2018-01-26:
use STM's official Core, they have better support for Nucleo boards. My repo is more orientated to generic / unbranded or home made boards 

Delete my files and use the JSON file as described here

https://github.com/stm32duino/Arduino_Core_STM32/blob/master/README.md

artbody, 2018-01-26:
i gave your code a try, because of SPI_2 implentation.
Nice work, i hope it'll be implemented in the official core 
so i switch back to the official release

rogerclarkmelbourne, 2018-01-26:
I'm not sure if SPI_2 is implemented in the Official core. 

You could try posting an issue on their core if its not implemeted

fpistm, 2018-01-26:
SPI is implemented.

artbody, 2018-01-28:
Ok thx 
special to fpistm
that#s the thing i was looking for, but didn't find this instruction.
by using the newest version of the official core
```
SPI.setMOSI(PB15);
SPI.setMISO(PB14);
SPI.setSCLK(PB13);
//..............^  this has to be setSCLK insted of setSCK
SPI.begin();
```
it compiles yes -> but doesn't work -> but that's not an issue for this core

rogerclarkmelbourne, 2018-01-28:
I'd need to look at what the official STM core does inside SPI.setMOSI etc

AFIK, there are only 2 possible pins for SPI2 MOSI etc, the normal pin and the alternate pin.
The same applies for SPI1, 

In this core; the SPI class has a function called setModule(), which allows you to designate whether "SPI" means SPI interface 1 or SPI interface 2 on the MCU

e.g. if you want use a library which is hard coded to use "SPI", you can call SPI.setModule(2), just prior to calling any functions relating to that library, and it re-configures "SPI" to mean SPI interface 2

But I don't have functions inside the SPI class to use the "alternate" pins.

I think setMOSI etc, could be missleading on devices like the F103, because with hardware SPI, you can't set any arbitrary pins.
Note Some MCU's do allow this sort of reconfiguration e.g. I know the Nordic nRF5x series can reconfigure virtually any pin to any function e.g. SPI, I2C or Serial, but this is not possible on the F103 or perhaps none of the STM32Fx series

fpistm, 2018-01-28:
@rogerclarkmelbourne 
As you said by default, core define a SPI instance.
The setXXXX allow to change default pin to use for this SPI instance before the begin.
Then the begin check if pins have the capabilities.
STM core knows all Peripherals linked to a pin, it's a part of the pin mapping.
ex:
https://github.com/stm32duino/Arduino_Core_STM32/blob/113f9e6368b5a093a4c71ba02bf8d820cd61f05e/variants/NUCLEO_F103RB/PeripheralPins.c#L204

all infos are encoded on an uint32:
```
/*  STM PIN data as used in pin_function is coded on 32 bits as below
 *   [2:0]  Function (like in MODER reg) : Input / Output / Alt / Analog
 *     [3]  Output Push-Pull / Open Drain (as in OTYPER reg)
 *   [5:4]  as in PUPDR reg: No Pull, Pull-up, Pull-Donc
 *   [7:6]  Reserved for speed config (as in OSPEEDR), but not used yet
 *  [14:8]  Alternate Num (as in AFRL/AFRG reg)
 * [19:15]  Channel (Analog/Timer specific)
 *    [20]  Inverted (Analog/Timer specific)
 *    [21]  Analog ADC control - Only valid for specific families
 * [32:22]  Reserved
 */
```

rogerclarkmelbourne, 2018-01-28:
Hi Frederic

Unless I'm mistaken,   if I call SPI.setMOSI(PA0);  this wont work because PA0 is not an AF REMAP of MOSI on SPI 1

??????

rogerclarkmelbourne, 2018-01-28:
BTW. I didnt realise that the SPI pins can have AF REMAP applied individually 

I thought perhaps they all had to be changed at the same time.

But I think this is just my missunderstanding

fpistm, 2018-01-28:
Right, as PA0 have no SPI Peripheral.
I do not well understand your thought about:

> SPI pins can have AF REMAP applied individually
> I thought perhaps they all had to be changed at the same time.

When `SPI.begin()` is called all spi pins (MOSI/MISO/SCK/SS  set thanks the constructor or using the setXXX methods) are configured.
https://github.com/stm32duino/Arduino_Core_STM32/blob/113f9e6368b5a093a4c71ba02bf8d820cd61f05e/cores/arduino/stm32/spi_com.c#L282

rogerclarkmelbourne, 2018-01-29:
Hi Frederic

Can user call 

SPI.setMOSI(PB15);

But not all 

SPI.SPI.setMISO()

etc ?

I think if I implemented 

SPI.setMOSI(PIN_NAME);  
etc

I would have lots of bug reports, as people who have no experience with STM32 would think that they can use any PIN_NAME with MOSI, or MISO or CLK
 
:-(

fpistm, 2018-01-29:
Right, 

Note that, STM core also provided a constructor with pins params, it is the same issue:
`    SPIClass(uint8_t mosi, uint8_t miso, uint8_t sclk, uint8_t ssel = (uint8_t)NC);`
setXXX method has been added to be able to change pin used by the default SPI instance which is hard-coded in several libs. This allows some flexibility but right if user use a pin with no SPI capabilities this will not work ...
But some STM32 MCU could have up to 6 SPI peripherals (ex F7) and available across 15 pins.

rogerclarkmelbourne, 2018-01-29:
Hi Frederic

Thanks for confirming

BTW. I don't think adding a printf to warn the user would be solve the problem. 
This is a discussion in one of the threads about why print in libraries can be a problem (Sorry I can't remember which thread it is, but Ray posted and I posted recently.

I'm not sure there is a good solution for when the user tries an unsupported pin.

fpistm, 2018-01-29:
PEBCAK ;)

fpistm, 2018-01-29:
@artbody 
I've just tested all SPIx available by only changing pins to used thanks setXXX() with the Nucleo F103RB and it's OK.
To test it, I used the TFTBitmapLogo example.
By default, SPI instance uses SPI1 without AF (PA7/PA6/PA5)
Then I've tested SP1 remapped to PB5/PB4/PB3. (warning PB3 is linked to Serial wire JTAG configuration, you have to  disable it `pinF1_DisconnectDebug(PB_3);`
Then  I've tested SPI2 (PB15/PB14/PB13).

SPI pin must be changed before  `TFTscreen.begin();` which call `SPI.begin();`
```
void setup() {
  /* SPI1 remapped */
  /*
  pinF1_DisconnectDebug(PB_3);
  SPI.setMOSI(PB5);
  SPI.setMISO(PB4);
  SPI.setSCLK(PB3);
  */
  /* SPI2 */
  /*
  SPI.setMOSI(PB15);
  SPI.setMISO(PB14);
  SPI.setSCLK(PB13);
  */
  // initialize the GLCD and show a message
  // asking the user to open the serial line
  TFTscreen.begin();

```
Note: check the Errata sheets of the STM32F103, several issues are listed:
http://www.st.com/resource/en/errata_sheet/cd00190234.pdf

@rogerclarkmelbourne sorry for the pollution as not linked to your core.

rogerclarkmelbourne, 2018-02-12:
@fpistm 
No worries

closed 2018-02-12 by rogerclarkmelbourne

rtek1000, 2019-09-23:
If someone looks for SPI 1 remapping on Blue Pill (stm32f103c8t6):

https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/536?_pjax=%23js-repo-pjax-container#issuecomment-533972563

FlamingMoe85, 2020-01-10:
Hey there, 

I had issues too and found a first step for a solution.

(Medium)Long story short - It has to do with the baude rate. I´m using Rogers WS2812 lib (lot´s of thanks :>) and saw on an oszilloscope, that the pulses are two times too long. I searched the incuded spi files for locations dealing with the rate. 
In a first step I could get the WS2812 lib running on SPI_2 by hard coding 
spi_master_enable(_currentSetting->spi_d, **(spi_baud_rate)24/*_currentSetting->clockDivider*/**, (spi_mode)_currentSetting->dataMode, flags);
in Roger´s SPI.c

I guess it has to do with the APB1/2 clock domains which Roger cover´s in his SPI.c. May be, the reason is somethings get´s misscalculated or missmatched?

 

I´m using Rogers WS2812 library (many thanks) and wanted it to run on SPI_2 since I need the timer outputs.

stevstrong, 2020-01-11:
You can get the desired frequency on the SPI ports independent on the clock domains if you use
`SPI.begin(SPISettings(<freq_in_mhz>));`
or
`SPI.beginTransaction(SPISettings(<freq_in_mhz>));`