rogerclarkmelbourne/Arduino_STM32 #437: Compile error in current master branch

kbahey, 2018-02-01:
When using the current master, and choosing 'generic stm32f103v' board, which is actually the [CZ mini](http://www.stm32duino.com/viewtopic.php?f=28&t=490&start=50#p39409),   I get the following compile error:

../hardware/Arduino_STM32/STM32F1/variants/generic_stm32f103v/board.cpp:168:1: 
error: too many initializers for 'const stm32_pin_info [79]'
 };
 ^

This is because in:
../hardware/Arduino_STM32/STM32F1/variants/generic_stm32f103v/board/board.h
#define BOARD_NR_GPIO_PINS      79

While in: 
../hardware/Arduino_STM32/STM32F1/variants/generic_stm32f103v/board.cpp there are actually 80 lines. 

Not sure which one is correct, but one or the other needs to be fixed. 

Side issue: I registered at the STM32duino forum and I am not able to post there.

rogerclarkmelbourne, 2018-02-01:
New users of the forum Must post to the New Users section first. 
This measure had to be put in place to stop AI SpamBots registering and posting all sorts of rubbish to random topics

rogerclarkmelbourne, 2018-02-01:
BTW. 
I just tested your bug, and I can confirm it is happening.

I'm not sure how that crept in. 

Perhaps one of the PR's removed one of the pins, and I didn't notice, as occasionally PR's change files that are nothing to do with the PR :-(

kbahey, 2018-02-01:
Changing the 79 to 80 (to match the number of lines) works for me.

But can't tell if there are duplicates in the 80 lines or what.

rogerclarkmelbourne, 2018-02-01:
I agree

That number needs to be changed

I posted a comment to the PR that caused the issue, and perhaps @victorpv  knows why the other file didnt get updated

Its my mistake. I probably assumed this PR was so simple, I'd approve and not fire up the Arduino IDE and test it

rogerclarkmelbourne, 2018-02-01:
Should now be fixed by

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/42a5e9f9073e1f3b2a479abc4145a32b8e8c9339

Please retest.

victorpv, 2018-02-03:
Sorry, seems like my PR produced the error.
I normally compile with sloeber, and was not throwing me an error, so I never noticed BOARD_NR_GPIO_PINS  had to be modified (my local copy was still at 79).

I need to reinstall Arduino IDE so next time I test my code in Arduino before sending a PR.

closed 2018-02-12 by rogerclarkmelbourne