rogerclarkmelbourne/Arduino_STM32 #447: Nucleo-F103RB Serial port typo

RickKimball, 2018-02-20:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/variants/nucleo_f103rb/board.cpp#L256

Line 256 is wrong:
```
   DEFINE_HWSERIAL(Serial, 3);// Use HW Serial 2 as "Serial"
```
It should be:
```
   DEFINE_HWSERIAL(Serial, 2);// Use HW Serial 2 as "Serial"
```
The current way it is configured makes Serial use USART3.  This is bad because, the STLINK/V2.1 virtual com port is wired to USART2.  This makes it possible to use the normal examples that use Serial to work properly without having any native USB or having to use an external usb serial dongle.

madias123, 2018-03-29:
Rick: This is a historic thing: As I setup the board files for the nucleo I had a goal: To be as most as possible pin compatible with the arduino UNO, so you can most of the "shields".  After years (I don't use the RB103, just crack the STLink 2.1 as only using part of it) I would say: Let it as it is, because you won't get a 1:1 pin compatibility with the UNO. In my opinion ST messed this totally up. (Why "Arduino" headers when they aren't really compatible?) There is a *doc file in the nucleo folder where I had written everything about the remappings.

RickKimball, 2018-04-13:
@madias123 I think you should be able to open a new Nucleo F103RB and use it with the stlink v2-1's virtual com port.  If you don't use it any more I don't understand why you want to deprive the rest of us from just working.

No matter, this is a 5 second fix. I'll just use my personal version.  At least this is here as an issue that someone else can find.

closed 2018-04-13 by RickKimball