rogerclarkmelbourne/Arduino_STM32 #448: Nucleo-32 Arduino Compile Problems

joedarock, 2018-02-20:
I'm running Arduino 1.8.3 and have installed the cores for the Nucleo boards. I can successfully compile a simple Blink program and load it into the STM32F303 board. However, when I try to compile almost anything else, I get an error message that just says:

The system cannot find the path specified.
exit status 1
Error compiling for board Nucleo-32.

With verbose mode on, I get the following additional info:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Joe Rocci\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Joe Rocci\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Joe Rocci\Documents\Arduino\libraries -fqbn=STM32:stm32:Nucleo_32:pnum=NUCLEO_F303K8,upload_method=STLink,xserial=generic,usb=none,opt=osstd -ide-version=10803 -build-path C:\Users\JOEROC~1\AppData\Local\Temp\arduino_build_269521 -warnings=none -build-cache C:\Users\JOEROC~1\AppData\Local\Temp\arduino_cache_536312 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM32Tools.path=C:\Users\Joe Rocci\AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\2017.9.22 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Joe Rocci\AppData\Local\Arduino15\packages\STM32\tools\arm-none-eabi-gcc\6-2017-q2-update -prefs=runtime.tools.CMSIS.path=C:\Users\Joe Rocci\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -verbose C:\Users\Joe Rocci\Google Drive\STM32_Version\ubitx_20\ubitx_20.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Joe Rocci\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Joe Rocci\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Joe Rocci\Documents\Arduino\libraries -fqbn=STM32:stm32:Nucleo_32:pnum=NUCLEO_F303K8,upload_method=STLink,xserial=generic,usb=none,opt=osstd -ide-version=10803 -build-path C:\Users\JOEROC~1\AppData\Local\Temp\arduino_build_269521 -warnings=none -build-cache C:\Users\JOEROC~1\AppData\Local\Temp\arduino_cache_536312 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM32Tools.path=C:\Users\Joe Rocci\AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\2017.9.22 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Joe Rocci\AppData\Local\Arduino15\packages\STM32\tools\arm-none-eabi-gcc\6-2017-q2-update -prefs=runtime.tools.CMSIS.path=C:\Users\Joe Rocci\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -verbose C:\Users\Joe Rocci\Google Drive\STM32_Version\ubitx_20\ubitx_20.ino
Using board 'Nucleo_32' from platform in folder: C:\Users\Joe Rocci\AppData\Local\Arduino15\packages\STM32\hardware\stm32\2018.1.18
Using core 'arduino' from platform in folder: C:\Users\Joe Rocci\AppData\Local\Arduino15\packages\STM32\hardware\stm32\2018.1.18
cmd /c "if not exist C:\Users\Joe Rocci\Google Drive\STM32_Version\ubitx_20/build_opt.h mkdir C:\Users\JOEROC~1\AppData\Local\Temp\arduino_build_269521\sketch & type NUL > C:\Users\JOEROC~1\AppData\Local\Temp\arduino_build_269521/sketch/build_opt.h"

Any suggestions?
Joe

fpistm, 2018-02-21:
Hi Joe,
Your issue is not related to @rogerclarkmelbourne core (Arduino_STM32) but this one:
https://github.com/stm32duino/Arduino_Core_STM32
It is a know issue and will be fixed to the next release.
See https://github.com/stm32duino/Arduino_Core_STM32/pull/204
br

joedarock, 2018-02-21:
FrederickThanks. I reloaded the STM hardware cores and it seems to be working now.
Joe

https://github.com/stm32duino/Arduino_Core_STM32

See stm32duino/Arduino_Core_STM32#204

br

joedarock, 2018-02-22:
Roger

Sorry for re-posting the same issue. I came back and looked for my original post but couldn't find it.

Is there a temporary workaround?

Joe

fpistm, 2018-02-23:
@joedarock,
This is not an issue of @rogerclarkmelbourne core.
the issue come from the fact you have space in Arduino path.
In the PR i've mentionned before there is the fix in platform.txt.
https://github.com/stm32duino/Arduino_Core_STM32/pull/204/commits/87ed5e35c39b787ac2d2081be67cec0de800c304#diff-d638a47d1ca0e2a5585fb34c7d7fb0f9R87
apply it to your platform.txt.

joedarock, 2018-02-23:
Replacing the Windows recipe with the one on GitHub fixed it...thanks

closed 2018-02-23 by joedarock