rogerclarkmelbourne/Arduino_STM32 #449: linker unable to reference _sbrk, STM32F103C8

MichaelFay, 2018-02-23:
When trying to compile the base sketch I am receiving the following error: 

/Users/michaelfay/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/libc.a(lib_a-sbrkr.o): In function `_sbrk_r':
sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'
collect2: error: ld returned 1 exit status
*** [.pioenvs/genericSTM32F103C8/firmware.elf] Error 1

rogerclarkmelbourne, 2018-02-23:
I see you are using Platfomio.

This repo is designed to work with the official Arduino IDE (this made clear in the Readme.md file)

Please contact the Platformio team about this.

closed 2018-02-23 by rogerclarkmelbourne

JiriBilek, 2018-02-23:
Michael,
look at http://stm32duino.com/viewtopic.php?f=41&t=2434&start=80#p33786
Maybe it helps you. I had similar problems with not liked libraries in Sloeber IDE recently.

MichaelFay, 2018-02-25:
It looks like when you use a type not specified by c, it attempts to build in the arm libraries. For example:

String foo()
{
return "String"
}

JiriBilek, 2018-02-25:
@MichaelFay , my problems started with the String library, too. This library uses malloc to allocate room for strings and malloc calls _sbrk. Have you tried to edit platform.txt as is suggested in the forum post?

MichaelFay, 2018-02-26:
It appears that this was just brought up in the platform IO thread. For anyone else that rounds into this problem the link to the discussion thread is [here](https://community.platformio.org/t/error-on-stm32-arduino-and-sprintf/3584).

ivankravets, 2018-03-02:
Sorry, that was our typo and we already fixed. Please navigate to `PIO Home > Platforms > Updates` and update ST STM32 dev/platform.