rogerclarkmelbourne/Arduino_STM32 #453: HardwareTimer::setOverflow() freezes timer on Black Pill

DeshmukhMalhar, 2018-03-06:
[t_enum_base.txt](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/1785282/t_enum_base.txt)

DeshmukhMalhar, 2018-03-07:
It seems that it works for overflow value of 255, but whay does not work for 1023 ?

stevstrong, 2018-03-07:
Replace
`  while (!Serial.available());
`
with:
`  while (!Serial); delay(10);
`
and it should work.

DeshmukhMalhar, 2018-03-09:
Okay . Will try that.
Thanks

sweetlilmre, 2018-03-09:
Out of interest, why does this change resolve the issue?

stevstrong, 2018-03-09:
That is the question for which I am also looking the answer.
I am investigating the issue.

stevstrong, 2018-03-10:
The good news: it does not freezes the MCU, only the USB serial will not output anything in the monitor.
You can insert in the loop a LED blinky part which works fine.
It looks like a race condition in the USB CDC driver.

stevstrong, 2018-03-11:
It looks like it is an issue related to Arduino IDE serial monitor.
If I use another serial monitor (for example CoolTerm) all messages are printed.

Please try to use another serial monitor software to acknowledge my findings.

If you stick to Arduino serial monitor then, as a workaround, you should use at least a delay(1);

I think this can be closed.

DeshmukhMalhar, 2018-03-14:
Sorry for late reply.
It I tried with brightness of the LED, not with Serial port.
Brightness of LED does not change/reduce.
I'll provide more details once I get home.

I think this issue is related with https://github.com/Microsoft/vscode-arduino and VScode.
Please Wait till I confirm that.
Thank you

rogerclarkmelbourne, 2018-05-23:
Closed due to no activity in the last 8 weeks. Please reopen if necessary

closed 2018-05-23 by rogerclarkmelbourne