rogerclarkmelbourne/Arduino_STM32 #454: troubleshhoting the "STM32duino bootloader" upload method - Linux Mint 18.3 (64bit)

lucky62, 2018-03-10:
Hello,
I have burned the bootloader (generic_boot20_pc13.bin) to the my "blue pill" board, but I cannot load the sketch using  "STM32duino bootloader" method.

bootloader is loaded successfully - I see these messages when connected to ttyACM0:
```
Congratulations, you have installed the STM32duino bootloader
See https://github.com/rogerclarkmelbourne/STM32duino-bootloader

For more information about Arduino on STM32
and http://www.stm32duino.com

```

I already have the 1k8 resistor between PA12 and 3.3V.
I have increased upload-reset value in the maple_upload script.

Still every time I got the result:
`dfu-util: No DFU capable USB device available`

Kernel messages are showing that first is connected device 1EAF:0003, then it is disconnected and new device 1EAF:0004 is connected.

```
Mar 10 09:33:42 T470 kernel: [ 6236.657072] usb 1-2: new full-speed USB device number 47 using xhci_hcd
Mar 10 09:33:43 T470 kernel: [ 6236.802412] usb 1-2: New USB device found, idVendor=1eaf, idProduct=0003
Mar 10 09:33:43 T470 kernel: [ 6236.802416] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 10 09:33:43 T470 kernel: [ 6236.802419] usb 1-2: Product: Maple 003
Mar 10 09:33:43 T470 kernel: [ 6236.802421] usb 1-2: Manufacturer: LeafLabs
Mar 10 09:33:43 T470 kernel: [ 6236.802423] usb 1-2: SerialNumber: LLM 003
Mar 10 09:33:44 T470 kernel: [ 6238.128071] usb 1-2: USB disconnect, device number 47
Mar 10 09:33:44 T470 kernel: [ 6238.424770] usb 1-2: new full-speed USB device number 48 using xhci_hcd
Mar 10 09:33:44 T470 kernel: [ 6238.566776] usb 1-2: New USB device found, idVendor=1eaf, idProduct=0004
Mar 10 09:33:44 T470 kernel: [ 6238.566781] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Mar 10 09:33:44 T470 kernel: [ 6238.566784] usb 1-2: Product: Maple
Mar 10 09:33:44 T470 kernel: [ 6238.566786] usb 1-2: Manufacturer: LeafLabs
Mar 10 09:33:44 T470 kernel: [ 6238.567750] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
```
lsusb:

```
Bus 002 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 138a:0097 Validity Sensors, Inc. 
Bus 001 Device 005: ID 04f2:b5ab Chicony Electronics Co., Ltd 
Bus 001 Device 004: ID 8087:0a2b Intel Corp. 
Bus 001 Device 003: ID 0e8f:00a7 GreenAsia Inc. 
Bus 001 Device 050: ID 1eaf:0004  
Bus 001 Device 007: ID 2386:310e  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

However Arduino IDE is using the 0003 value, in addition - 32bit dfu-util is used (from linux directory instead of linux64 directory) even I am on 64 bit linux (Mint 18.3):

```
Sketch uses 13084 bytes (19%) of program storage space. Maximum is 65536 bytes.
Global variables use 2832 bytes (13%) of dynamic memory, leaving 17648 bytes for local variables. Maximum is 20480 bytes.
/SHARED/SSD_DATA/00_DevTools/00_Arduino/arduino-1.8.5/hardware/Arduino_STM32-master/tools/linux/maple_upload ttyACM0 2 1EAF:0003 /tmp/arduino_build_325682/STM32F103_Test.ino.bin 
dfu-util 0.8
dfu-util: Invalid DFU suffix signature

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
Waiting for /dev/ttyACM0 serial...Done
```
So I did some modifications in maple_upload script (hard-coded corrections and some debug prints) - but no success:

```
Sketch uses 13084 bytes (19%) of program storage space. Maximum is 65536 bytes.
Global variables use 2832 bytes (13%) of dynamic memory, leaving 17648 bytes for local variables. Maximum is 20480 bytes.
/SHARED/SSD_DATA/00_DevTools/00_Arduino/arduino-1.8.5/hardware/Arduino_STM32-master/tools/linux/maple_upload ttyACM0 2 1EAF:0003 /tmp/arduino_build_252422/STM32F103_Test.ino.bin 
port:ttyACM0, altID:2, usbID:1EAF:0004, binfile:/tmp/arduino_build_252422/STM32F103_Test.ino.bin, fullpath:/dev/ttyACM0
DFU_UTIL=/SHARED/SSD_DATA/00_DevTools/00_Arduino/arduino-1.8.5/hardware/Arduino_STM32-master/tools/linux64/dfu-util/dfu-util
======== before DFU
No DFU capable USB device found
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x1eaf product = 0x0004
======== after DFU
Waiting for /dev/ttyACM0 serial...Done
```

Last point - seems that something (ModemManager?) is blocking the ttyACM0 port for quite a long time (cca 17 seconds) after usb is connected. I  can connect to this port only after last ModemManager message:
```
Mar 10 10:49:21 T470 kernel: [10775.669959] usb 1-2: new full-speed USB device number 56 using xhci_hcd
Mar 10 10:49:21 T470 kernel: [10775.669959] usb 1-2: new full-speed USB device number 56 using xhci_hcd
Mar 10 10:49:22 T470 kernel: [10775.815718] usb 1-2: New USB device found, idVendor=1eaf, idProduct=0004
Mar 10 10:49:22 T470 kernel: [10775.815723] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Mar 10 10:49:22 T470 kernel: [10775.815726] usb 1-2: Product: Maple
Mar 10 10:49:22 T470 kernel: [10775.815729] usb 1-2: Manufacturer: LeafLabs
Mar 10 10:49:22 T470 kernel: [10775.816674] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
Mar 10 10:49:22 T470 kernel: [10775.815718] usb 1-2: New USB device found, idVendor=1eaf, idProduct=0004
Mar 10 10:49:22 T470 kernel: [10775.815723] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Mar 10 10:49:22 T470 kernel: [10775.815726] usb 1-2: Product: Maple
Mar 10 10:49:22 T470 kernel: [10775.815729] usb 1-2: Manufacturer: LeafLabs
Mar 10 10:49:22 T470 kernel: [10775.816674] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
Mar 10 10:49:22 T470 mtp-probe: checking bus 1, device 56: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
Mar 10 10:49:22 T470 mtp-probe: bus: 1, device: 56 was not an MTP device
Mar 10 10:49:28 T470 colord[1090]: # ERROR: snmp_open("udp6:[ff02::1]") failed, Address family not supported by protocol
Mar 10 10:49:39 T470 ModemManager[984]: <info>  Creating modem with plugin 'Generic' and '1' ports
Mar 10 10:49:39 T470 ModemManager[984]: <warn>  Could not grab port (tty/ttyACM0): 'Cannot add port 'tty/ttyACM0', unhandled serial type'
Mar 10 10:49:39 T470 ModemManager[984]: <warn>  Couldn't create modem for device at '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2': Failed to find primary AT port

```
So what can I do next?..

lucky62, 2018-03-10:
Regarding the ModemManager issue I found that the following flag is in udev rules in linux directory (32bit), but not in linux64:
ENV{ID_MM_DEVICE_IGNORE}="1"

lucky62, 2018-03-10:
Hello, I have created the small script which is testing the delay parameter for upload-reset utility:
Script is using the dfu-util with parameter -l which is showing the DFU devices.
```
PORT="/dev/ttyACM0"
for ((i=1;i<=50;i++)); do
	let "DELAY = 50 * i + 1500"
	echo "DELAY = $DELAY"
	/SHARED/SSD_DATA/00_DevTools/00_Arduino/arduino-1.8.5/hardware/Arduino_STM32-master/tools/linux/upload-reset $PORT $DELAY
	/home/lucky62/.arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1/dfu-util -l
	
	COUNTER=0
	while [ ! -r $PORT ] && ((COUNTER++ < 40)); do
		sleep 0.1
	done
done
```
The following output of script shows the minimal and maximal value of upload-reset delay.
When delay is out of this range no DFU device is shown.
Script output is cutted:

```
$ ./linux/upload_reset_test.sh
DELAY = 50
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DELAY = 100
setDTR(): TIOCMGET: Input/output error
setRTS(): TIOCMGET: Input/output error
setDTR(): TIOCMGET: Input/output error
setDTR(): TIOCMGET: Input/output error
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DELAY = 150
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
```
First DFU devices was detected at 650ms delay...
```
DELAY = 650
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [1eaf:0003] ver=0201, devnum=119, cfg=1, intf=0, path="1-2", alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000", serial="LLM 003"
Found DFU: [1eaf:0003] ver=0201, devnum=119, cfg=1, intf=0, path="1-2", alt=1, name="STM32duino bootloader v1.0  Upload to Flash 0x8005000", serial="LLM 003"
Found DFU: [1eaf:0003] ver=0201, devnum=119, cfg=1, intf=0, path="1-2", alt=0, name="STM32duino bootloader v1.0  ERROR. Upload to RAM not supported.", serial="LLM 003"
DELAY = 700
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
[... 59 lines cut ...]
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device

Download	[                         ]   0%            0 bytes
Download	[=                        ]   7%         1024 bytes
Download	[===                      ]  14%         2048 bytes
Download	[=====                    ]  21%         3072 bytes
Download	[=======                  ]  29%         4096 bytes
Download	[=========                ]  36%         5120 bytes
Download	[==========               ]  43%         6144 bytes
Download	[============             ]  50%         7168 bytes
Download	[==============           ]  58%         8192 bytes
Download	[================         ]  65%         9216 bytes
Download	[==================       ]  72%        10240 bytes
Download	[===================      ]  79%        11264 bytes
Download	[=====================    ]  87%        12288 bytes
Download	[=======================  ]  94%        13068 bytes
Download	[=========================] 100%        13068 bytes
Download done.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
Waiting for /dev/ttyACM0 serial...Done
```

closed 2018-03-10 by lucky62

rogerclarkmelbourne, 2018-03-10:
Community support is via www,stm32duino.com

SalimF, 2020-02-09:
this may be not directly related to the issue , but all uploading complaining errors I had were related to USB state of the board , it have to id number one ends with :0004 and the other with :0003 which is the default state after connecting it , after few second (I didn't maybe 30 or 60 ) device id changed to :0004 which will not allow you to upload any sketch .

rogerclarkmelbourne, 2020-02-09:
This is by design. 

The bootloader is on ID 0003 but if you install a sketch with USB Serial enabled (most build options have the USB Serial enabled), the sketch will appear as ID 0004

The applications in the tools folder (the maple loader), sends a special serial sequence to the sketch which causes the sketch to reboot the MCU into the bootloader.