rogerclarkmelbourne/Arduino_STM32 #455: Symbol collision

arpruss, 2018-03-12:
There is a collision between 
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4021c32fdb809f16a61c073d04e1d8c2c487a767/STM32F1/system/libmaple/usb/usb_lib/usb_type.h
and
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/c5133eba8ff6babf058c61cb55508af91c620c5a/STM32F1/libraries/Wire/utility/WireBase.h
both of which define a SUCCESS symbol. The two symbols, however, are defined to have the opposite values. 

I suggest prefixing the values in usb_type.h with USB_, like was already done in the STM32F4 core.

rogerclarkmelbourne, 2018-03-12:
When do you get an error using these ?

But I agree that adding prefixing the USB version seems the best solution

arpruss, 2018-03-12:
I had an error when compiling code that included both my USB library header and SoftWire.h.

stevstrong, 2018-03-12:
It seems that the USB does not use that symbol, I couldn't find it in the source code.
What I have seen was USB_SUCCESS overall.

arpruss, 2018-03-12:
Actually, we can't rename to USB_SUCCESS, because that symbol is used with the opposite value here:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4f753fc14e7fee0d0c99ab3dd127d193ad2164f8/STM32F1/system/libmaple/usb/usb_lib/usb_core.h

Looks like the SUCCESS symbol is used to define the ErrorStatus enum type, but the ErrorStatus type doesn't seem to be used anywhere in the code. So a nice fix would be just to remove the whole ErrorStatus enum.

victorpv, 2018-04-12:
If we can't remove ErrorStatus, could we change the members of the enum so USB_SUCCESS is used?

typedef enum {USB_SUCCESS,  USB_ERROR} ErrorStatus;

But if removing it doesn't break anything, then let's remove it.

stevstrong, 2018-04-12:
I agree with removal.

rogerclarkmelbourne, 2018-04-12:
I searched for the enum type and could not find anything but the definition

https://github.com/rogerclarkmelbourne/Arduino_STM32/search?utf8=%E2%9C%93&q=ErrorStatus&type=

So I commented it out and did a test build and it compiled fine

As far as I an tell the definition 

`typedef enum { ERROR = 0, SUCCESS  = !ERROR} ErrorStatus;`

is redundant

victorpv, 2018-04-12:
Probably a left over since that code was taken out of the spl.
Good to me, I'll test when I get a chance and will post if any issue.

rogerclarkmelbourne, 2018-04-13:
I've removed it in this commit

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/479cfc1fb4a39cb8398b309805eef93ba33e76fe

Let me know if that fixes it.

closed 2018-04-13 by rogerclarkmelbourne