rogerclarkmelbourne/Arduino_STM32 #456: Lcd7920_STM library not working with func ::bitmap

blygnguyen, 2018-03-12:
Hi.
I try to use lcd.bitmap with some bitmap image. but It displays with the same thing.

stevstrong, 2018-03-12:
Please post a simple example to reproduce the problem.
Btw, what means

> It displays with the same thing

?

blygnguyen, 2018-03-12:
I upload with code lcd.bitmap(0,0,10,30,myLogo); or other bitmap image. But in my gLCD is display the same thing. It not my logo image, it display like broken image. Sorry for my first time use github.

stevstrong, 2018-03-12:
Try to replace the bitmap function with this one:
```
void Lcd7920::bitmap(uint8_t x0, uint8_t y0, uint8_t width, uint8_t height, const uint8_t * data)
{
  for (uint8_t r = 0; r < height && r + y0 < numRows; ++r)
  {
    uint8_t *p = image + (((r + y0) * (numCols/8)) + (x0/8));
    uint16_t bitMapOffset = r * (width/8);
    for (uint8_t c = 0; c < (width/8) && c + (x0/8) < numCols/8; ++c)
    {
      *p++ = data[bitMapOffset++];
    }
  }
  if (x0 < startCol) startCol = x0;
  if (x0 + width > endCol) endCol = x0 + width;
  if (y0 < startRow) startRow = y0;
  if (y0 + height > endRow) endRow = y0 + height;
}
```

blygnguyen, 2018-03-13:
Thank you very much. 
That code is working. My image was displayed. 
Thanh you. 👍 👍 👍

closed 2018-03-13 by blygnguyen

eme-lyn, 2018-05-15:
hi :) I'm trying to use le bitmap function too and I have the same problem as blygnguyen my image display like broken and I change the part of the code that stevstrong suggested but it didn't work :( I'm using a STM32F103RB if you can suggest me something I will be grateful. 
thanks.

eme-lyn, 2018-05-16:
Ow :(  and i have another question, do you know how to clean the LCD? Because I'm using the function LCD.clear and I'm adding a delay but didn't work :(

blygnguyen, 2018-05-16:
ohh. lcd.clear is working too.

eme-lyn, 2018-05-16:
Could I see your code please? 😧😇 If you have one example of that.
Thanks :)

eme-lyn, 2018-05-16:
Also I tried with lcd.fillbox (0,0, maxwidth, maxhigh, PixelClear) but it didn't work 
I try to display a message  and for a certain time I what to clean the display.
Thanks for read me :3