rogerclarkmelbourne/Arduino_STM32 #463: RTClock example is not working.

blygnguyen, 2018-03-16:
The STM32f1 stop working after first call the RTC func. need some help!!

BlackBrix, 2018-03-16:
I think it's better to use the forum for such kind of questions --> http://stm32duino.com

blygnguyen, 2018-03-16:
I still can not solve the problem. my problem like this questions
[http://stm32duino.com/viewtopic.php?f=28&t=2724](url)
but i think my Blue Pill have 32768khz crystal. my board: [http://wiki.stm32duino.com/index.php?title=Blue_Pill](url)
i test rtclock example in [https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/RTClock/examples](url)
and only `RTClock rt (RTCSEL_HSE); ` is working. 
`RTClock rt (RTCSEL_LSE); ` is not.

stevstrong, 2018-03-16:
Please use the forum for such questions!
Does this help: http://stm32duino.com/viewtopic.php?f=3&t=2937&p=39134&hilit=blue+pill+rtc#p39134 ?

closed 2018-03-16 by blygnguyen