rogerclarkmelbourne/Arduino_STM32 #482: arduino not installing

ekaggrat, 2018-04-09:
i installed arduino due boards and then copied the arsuino_stm32 folder to the hardware folder.. but the boards dont show up and i am getting the following error

Could not find boards.txt in C:\Users\esk\Documents\Arduino\hardware\Arduino_STM32\cores. Is it pre-1.5?
Could not find boards.txt in C:\Users\esk\Documents\Arduino\hardware\Arduino_STM32\libraries. Is it pre-1.5?
Could not find boards.txt in C:\Users\esk\Documents\Arduino\hardware\Arduino_STM32\system. Is it pre-1.5?
Could not find boards.txt in C:\Users\esk\Documents\Arduino\hardware\Arduino_STM32\variants. Is it pre-1.5?
WARNING: Error loading hardware folder C:\Users\esk\Documents\Arduino\hardware\Arduino_STM32
  No valid hardware definitions found in folder Arduino_STM32.
Could not find boards.txt in C:\Users\esk\Documents\Arduino\hardware\Arduino_STM32\cores. Is it pre-1.5?
Could not find boards.txt in C:\Users\esk\Documents\Arduino\hardware\Arduino_STM32\libraries. Is it pre-1.5?
Could not find boards.txt in C:\Users\esk\Documents\Arduino\hardware\Arduino_STM32\system. Is it pre-1.5?
Could not find boards.txt in C:\Users\esk\Documents\Arduino\hardware\Arduino_STM32\variants. Is it pre-1.5?
WARNING: Error loading hardware folder C:\Users\esk\Documents\Arduino\hardware\Arduino_STM32
  No valid hardware definitions found in folder Arduino_STM32.

ekaggrat, 2018-04-09:
got it

closed 2018-04-09 by ekaggrat

rogerclarkmelbourne, 2018-04-09:
Its not installed correctly. you are missing part one level of the hierarchy.