rogerclarkmelbourne/Arduino_STM32 #487: Issue in PR #441

edogaldo, 2018-04-10:
As per topic http://stm32duino.com/viewtopic.php?f=2&t=3480&sid=4ce86bcf8df7ba9701ae6e4f269cbb58, this PR introduces a bug: it modifies function "rtc_set_count()" instead of function "rtc_get_count()" as declared.

KenWillmott, 2018-04-10:
Why "instead of"? It seems to me like the get method also must use a non-atomic access.

edogaldo, 2018-04-10:
In your first comment of #441 you said "_Modify **rtc_get_count()** and rtc_get_divider() to fix non-atomic accesses to RTC counter registers_" but then in your code you modified **rtc_set_count()**.
Furthermore in **rtc_set_count()** you are not updating the registers anymore.
This was not intended behaviour, correct?

edogaldo, 2018-04-13:
Fixed with #488 merge

closed 2018-04-13 by edogaldo