rogerclarkmelbourne/Arduino_STM32 #490: Bug in rtc_set_count(uint32 value) in rtc_util.c

asdrubalfuentes, 2018-04-11:
The call to this function indirectly do not change the registers CNTH and CNTL of RTC, because the "value" parameter pased do not have effect in thats registers, the original function look so:

void rtc_set_count(uint32 value) {
	uint32 h, l;
	rtc_clear_sync();
	rtc_wait_sync();
	rtc_wait_finished();
	rtc_enter_config_mode();
	do {
		h = RTC->regs->CNTH & 0xffff;
		l = RTC->regs->CNTL & 0xffff;
	} while (h ^ (RTC->regs->CNTH & 0xffff));
	rtc_exit_config_mode();
	rtc_wait_finished();
}
Obviously 'h' and 'l' are not in function of "value" variable, but Registers CNTH and CNTL respectively, for this reason that registers are not changed at the call to this procedure.

stevstrong, 2018-04-11:
Duplicate of #488

edogaldo, 2018-04-13:
Fixed with #488 merge

asdrubalfuentes, 2018-04-13:
Yes i see the changes made, thanks.

closed 2018-05-10 by rogerclarkmelbourne