rogerclarkmelbourne/Arduino_STM32 #500: Rename FreeRTOS to prevent library conflict

bratoff, 2018-04-16:
Version 7.0.1 of FreeRTOS for STM32duino should be renamed from FreeRTOS to FreeRTOS701 so that it does not conflict with the mainstream version of FreeRTOS.

rogerclarkmelbourne, 2018-05-18:
@victorpv 
You are the FreeRTOS expert

Does this sound OK to you ?

victorpv, 2018-05-19:
Renaming it should be fine, and make it easier to see that's the 7.0.1 version (the oldest that we have). 
I see no problem on that.

Now, I don't see where is the conflict with the mainstream version. 
We have the 9.0.0 version in another folder called FreeRTOS 900.
@bratoff can you provide more detail on what conflict you are trying to avoid?

bratoff, 2018-05-19:
@victorpv - There appears to be a naming conflict with version 10.0.0-10 which causes that version to always show "needs update" when the STM32duino core is active.  It must be the un-versioned FreeRTOS  7.0.1 causing it because the "needs update" goes away if a non-STM32duino board is selected.  (I have about 8 different board families configured in my IDE.)

victorpv, 2018-05-21:
@bratoff a couple of questions:
Did you test renaming the 7.0.1 folder?
What IDE are you using?
Where did you get the FreeRTOS 10 library from?

fpistm, 2018-05-21:
@victorpv  probably this one:
https://github.com/feilipu/Arduino_FreeRTOS_Library
But not applicable for STM32, anyway, I think his concern is about to have both libraries installed (one for AVR and one for STM32).
And the mainstream have the same name (`name=FreeRTOS`) than the one from Roger.
https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/FreeRTOS
This causing issue with library updater as folder have the same name.
There is no issue at build time as include names are different.

fpistm, 2018-05-21:
That's the point ;) Only rename the FrerRTOS folder to FreeRTOS701  ;)

rogerclarkmelbourne, 2018-05-21:
Fixed in https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/6a551bd9abcc632632d7c9e71a49e341a1600399

closed 2018-05-21 by rogerclarkmelbourne