rogerclarkmelbourne/Arduino_STM32 #506: HID Bootloader problems

denisn73, 2018-04-24:
I could not open the issue in "stm32-hid-bootloader" repository, so I'm writing here.
1. HID bootloader not recognizing:
-I rewrote and compiled the project with forced initialization of USB. When connected board to the USB, it as an unknown device.
-I changed button pin and the address of the user program to 0x08002000 and skip USB initialization, when it run the USB CDC is working normally.
2. In "board.txt" add Upload method "HID Bootloader", when compiling an error appears that file is not found:
>cannot open linker script file C:\Users\dev12\Documents\Arduino\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103c/ld/hid_bootloader.ld

rogerclarkmelbourne, 2018-04-24:
HID bootloader is not supported yet.

I did not include those files in the repo 

e.g.

See

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/variants/generic_stm32f103c/ld

It does not contain hid_bootloader.ld

Serasidis, 2018-04-25:
As Roger said, the HID bootloader is not supported yet because it is on development.
You can also try the test version from here:
https://github.com/Serasidis/STM32_HID_Bootloader
Extract the  Arduino_STM32_patch.zip file to your Arduino_STM32 folder. Do not forget to backup your files, before applying the patch.

denisn73, 2018-04-25:
Thank you very much. I'll try it =)
The issue can be closed

closed 2018-04-25 by denisn73

denisn73, 2018-04-25:
Serasidis, I compiled your bootloader, and this is also not recognized as a hid device =(
I modify Makefile::
>CFLAGS += -std=gnu99
>#LDFLAGS += -Wl,--print-memory-usage

Then compiled with Cygwin64.

Serasidis, 2018-04-25:
Which board do you use ? bluepill (http://wiki.stm32duino.com/index.php?title=Blue_Pill) ?
If not, what micro-controller has your board ?

denisn73, 2018-04-25:
I have my own board with the same circuitry that bluepill. 
The controller respectively STM32F103Cx.
I tested on several boards. DFU and CDC is working fine.
OS - Windows 10.
In the task manager the device is defined as an "unknown device" with an error code: "a device descriptor request failed"
The bootloader executable code is executed correctly (it normal works with PB2 input, PC13 output, check BKP register and check correct first byte of user programm), but something wrong with the USB HID