rogerclarkmelbourne/Arduino_STM32 #513: static var init with function blows up memory usage

GIPdA, 2018-05-05:
Something is wrong!
The following code compiles to a 75272 bytes code...

```
int foo() { return 1;}
void setup() {}
void loop() {
  static unsigned int test = foo();
}
```

Remove either 'static' or 'foo()' and is goes down to 15168 bytes. Some bad build options??

Using Generic STM32F103C (but same with a few other STM32 boards I tested), on macOS 10.11, Arduino 1.8.5. Arduino_STM32 from master. Any optimization mode.
Anybody reproducing this issue?

rogerclarkmelbourne, 2018-05-05:
Umm

I'll need to try some older commits and see when this problem has crept in

I wonder if its the new __weak changes to allow overloading of interrupt functions

rogerclarkmelbourne, 2018-05-05:
OK.

I've checked as far back as 2015, and its always done this.

There is a thread about this on www.stm32duino.com

Sorry, I don't have a link.

The problem is that the core supports RTOS's and they require thread save statics.

This adds a huge overhead to the codebase.

rogerclarkmelbourne, 2018-05-05:
OK
Try changing boards.txt and add

-fno-threadsafe-statics

to the end of one of the optimisation settings lines

e.g.

mapleMini.menu.opt.osstd.build.flags.optimize=-Os -fno-threadsafe-statics

GIPdA, 2018-05-06:
Just got the time to try, thanks for your fast reply!

The -fno-threadsafe-statics option fixed the issue*. But it doesn't seems to be a "good" solution... I mean, the root cause seems to be elsewhere, it can't be just thread safe statics init, right? What gets added due to that?

*However I get the warning:
`cc1: warning: command line option '-fno-threadsafe-statics' is valid for C++/ObjC++ but not for C [enabled by default]`
Is there a better place to put this option flag?

Thank you

RickKimball, 2018-05-06:
-fno-threadsafe-statics is a c++ only option. You should add it to the compiler.cpp.flags

GIPdA, 2018-05-06:
Ended up adding it to the flags in platform.txt. It applies globally for every board of this core, I don't know if boards.txt can add/override flags, or if there is any other better method.

stevstrong, 2019-09-16:
Possible duplicate of #444 
The users either avoid static var init within functions or apply the solution (-no-threadsafe-statics) provided by Roger here: https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/444#issuecomment-468088453
Should we apply the solution and let the FRETOS users handle this on their own?

rogerclarkmelbourne, 2020-05-18:
I don't think there is a perfect solution to this.

Its not a problem for most people. 

I'll just add a note to the readme.md

closed 2020-05-18 by rogerclarkmelbourne