rogerclarkmelbourne/Arduino_STM32 #529: USB Serial.print does not work when DTR == 1

me21, 2018-06-15:
`USBSerial::write` checks that `USBSerial` instance can be converted to `true` before sending: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4db3994d1c42a45eb7b8fd298347128765f09e7f/STM32F1/cores/maple/usb_serial.cpp#L122

`operator bool()` is implemented as: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4db3994d1c42a45eb7b8fd298347128765f09e7f/STM32F1/cores/maple/usb_serial.cpp#L216-L218

So, if DTR is high, it's impossible to send data to host.

It prevents me from using STM32F103CB as an USB-Serial converter for programming Espressif ESP32 chip which is installed on the same board. Various USB-Serial converters connect DTR pin to GPIO0 pin of ESP32 to boot it to bootloader.

rogerclarkmelbourne, 2018-06-16:
This is a known feature.

If you want to modify the code and submit a PR, the community will consider it.

me21, 2018-06-16:
Thanks. I have found [your thread](http://www.stm32duino.com/viewtopic.php?t=2345) from the last year exactly on this topic. I'll think about what can be done.

me21, 2018-08-24:
OK, then low DTR state is used as a flag to examine the incoming data; but why can't we send while DTR is low?

I think I can modify `usb_cdcacm_tx` function to overwrite the oldest data in the TX ring buffer. If you can think of a situation where it could lead to a data loss, I can make new behaviour configurable, with old behaviour being the default.

Or, as another preposition, I can change `size_t USBSerial::write(const uint8 *buf, uint32 len)`. Currently it always returns 0 if I'm not mistaken (https://github.com/rogerclarkmelbourne/ ... l.cpp#L119) because n variable is never modified. Let's make this function return the actual number of bytes written and let the user retry the transmission later if he wants. Also, a configurable TX timeout can be added to this function so that it doesn't get stuck in the loop forever.

Then we can remove the check for DTR from `USBSerial::operator bool()`.

As another thought, [Wikipedia](https://en.wikipedia.org/wiki/Data_Terminal_Ready#Null_modem_operation) says that

> On many operating systems, including Windows, the DTR line is held low while the serial port is unused and not being controlled by any applications.

Then we can make this check for DTR configurable - if DTR is low, then the port is probably closed, but if the user still wants to send something, he can. I, for one, needed to send while DTR is low because I used STM32 as USB-Serial bridge for ESP32 chip, and its bootloader sets DTR low and it needs bidirectional data exchange.

stevstrong, 2018-08-25:
It looks like we need to change anyway the USB Serial write to return the number of sent bytes.
A timeout to check whether the bytes were sent should be done on a higher app level, imho.

rogerclarkmelbourne, 2019-01-28:
These sort of changes are currently being undertaken

rogerclarkmelbourne, 2019-01-30:
See https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/5654b56fd6ab223bcc120eede50b3db8c2b18fb7

me21, 2019-01-30:
It would be nice if `operator bool()` honored `USB_SERIAL_REQUIRE_DTR` too. Otherwise, how can I tell if Serial can be used if I don't require DTR?

me21, 2019-01-30:
Yeah but `operator bool()` still checks DTR. It probably shouldn't do that if `USB_SERIAL_REQUIRE_DTR` is not defined?

stevstrong, 2019-01-30:
Sounds reasonable.

rogerclarkmelbourne, 2019-01-30:
I can change this, but it has undesired side effects. 

Specifically code that has

while(!Serial);

 to wait until the Arduino IDE serial terminal is open, does no do what it previous did.

I'll need to check this myself, but I found this answer on stackoverflow

https://arduino.stackexchange.com/questions/4556/what-does-the-line-while-serial-do-in-an-arduino-program

Which claims that the Serial boolean operator does nothing useful on AVR boards that use an external USB to Serial, but on boards with built in USB serial, it claims that these boards are waiting for the terminal to be opened,

In which case they would be waiting for DTR, which is what my code currently does

me21, 2019-01-31:
Then maybe just make USB_SERIAL_REQUIRE_DTR defined by default?
I guess this will not change current behaviour and has no side effects.
If user wants, he can edit sources and undefine that. Then `operator bool()` will not wait for DTR, but only for that user, because he presumably know what he's doing.

Or, if you don't like the idea of placing additional `#define` and requiring user to edit sources, it's possible to rename that define to `USB_SERIAL_NO_DTR` or something alike. It will have reverse logic, i.e. check DTR unless that symbol is defined. Then only compiler flag will be needed.

rogerclarkmelbourne, 2019-02-04:
The fix has been copied to the Master branch.

closed 2019-02-04 by rogerclarkmelbourne