rogerclarkmelbourne/Arduino_STM32 #535: #define false false

denisn73, 2018-07-03:
In USBComposite librarie
[image]

stevstrong, 2018-07-03:
Which file? Which line?

denisn73, 2018-07-03:
...\STM32F1\cores\maple/wirish.h:86:0: warning: "true" redefined [enabled by default]
#define true 0x1

...\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\lib\gcc\arm-none-eabi\4.8.3\include\stdbool.h:43:0: note: this is the location of the previous definition
 #define true true

...\STM32F1\cores\maple/wirish.h:87:0: warning: "false" redefined [enabled by default]
 #define false 0x0

...\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\lib\gcc\arm-none-eabi\4.8.3\include\stdbool.h:42:0: note: this is the location of the previous definition
 #define false false

In file included

                 from ...\STM32F1\libraries\USBComposite/usb_mass_mal.h:5:0,
                 from ...\STM32F1\libraries\USBComposite/USBMassStorage.h:7,
                 from ...\STM32F1\libraries\USBComposite/USBComposite.h:11,
                 from ...\STM32F1\libraries\USBComposite\USBHID.h:20:
                 from ...\STM32F1\libraries\USBComposite\USBHID.cpp:17:
                 from ...\STM32F1\libraries\USBComposite\USBHID.cpp:30:
                 from ...\STM32F1\libraries\USBComposite\USBMIDI.h:37,
                 from ...\STM32F1\libraries\USBComposite\USBMIDI.cpp:41:
                 from ...\STM32F1\libraries\USBComposite\USBMIDI.cpp:46: