rogerclarkmelbourne/Arduino_STM32 #538: Serial doesnt initialize when flashing with STLINK

coder137, 2018-07-08:
When using the Arduino library with platformio, this code below doesnt work

```
#include <Arduino.h>

#define PIN PB12

void setup() {
    // put your setup code here, to run once:
    pinMode(PIN, OUTPUT);

    Serial.begin(9600);
    while(!Serial);
}

void loop() {
    // put your main code here, to run repeatedly:
    digitalWrite(PIN, HIGH);
    delay(100);

    digitalWrite(PIN, LOW);
    delay(100);

    //Serial.println("Done");
}
```

The code gets stuck on `while(!Serial)`

Even while doing this

```
#include <Arduino.h>

#define PIN PB12

void setup() {
    // put your setup code here, to run once:
    pinMode(PIN, OUTPUT);

    Serial.begin(9600);
}

void loop() {
    // put your main code here, to run repeatedly:
    digitalWrite(PIN, HIGH);
    delay(100);

    digitalWrite(PIN, LOW);
    delay(100);

    Serial.println("Done");
}
``` 

The program runs, however no output on Serial monitor

**NOTE, I am flashing the code through STLINK, and have kept an FTDI board on my PA9 and PA10 pins**

This problem does not occur when I am flashing through Serial

```
upload_protocol = stlink //problems
upload_protocol = serial //no problems what so ever
```

I feel either I am doing something wrong, or it could be a bug with the STLINK tool.

Do let me know if I should raise the issue elsewhere
Thank you

justinschoeman, 2018-07-08:
Have you looked at the Wiki:
http://wiki.stm32duino.com/index.php?title=API#Serial_.26_USB_Serial
You should probably use Serial1 when uploading by STLink.

coder137, 2018-07-08:
Thank you very much for your quick response.

This worked for me. Ill close this issue now :)

closed 2018-07-08 by coder137