rogerclarkmelbourne/Arduino_STM32 #540: STM32F103C hang after run this sketch

okumaster, 2018-07-11:
What's wrong about my sketch, when I compile on uno, this works perfectly, but will crashing the stm. 

```c++
#include <Arduino.h>

float angka []= {2.344, -6391199, 106821679, -4.324};
void setup() 
{
  Serial.begin(9600);
}

void loop()
{
  unsigned long waktuawal;
  int waktuakhir; 
  waktuawal = millis();
  
  //Fungsi untuk mengkonversi dari bentuk gps ke standar latitude/longitude ie: (-6391199) => (-6.391199)
  String pre_lat, on_lat, post_lat, pre_lon, on_lon, post_lon;
  char lat_buff[9], lon_buff[10];
  pre_lat = String (angka[1]);
  pre_lon = String (angka[2]);
  pre_lat.toCharArray(lat_buff,9);
  pre_lon.toCharArray(lon_buff,10);
    if ((!pre_lat == 0) && (!pre_lon == 0)) { //Fungsi ini dijalankan apabila nilai lat&lon!=0
        
        Serial.println(lat_buff);
        for (int a=0; a<=8; a++) { //Konversi nilai lat
          if ((a < 2) || (a > 2)) {
            on_lat = String (lat_buff[a]);
            post_lat = post_lat+on_lat; 
          }else if (a == 2) {
            on_lat = String (lat_buff[a]);
            post_lat= post_lat+"."+on_lat;
          }
        }
     
        Serial.println(lon_buff);
        for (int b=0; b<=9; b++){ //Konversi nilai lon
          if ((b < 3) || (b > 3)) {
            on_lon = String (lon_buff[b]);
            post_lon = post_lon+on_lon;
          }else if (b == 3){
            on_lon = String (lon_buff[b]);
            post_lon = post_lon+"."+on_lon;            
          }
        }
    }
  Serial.println(post_lat);
  Serial.println(post_lon);
  Serial.flush();
  waktuakhir = waktuawal - millis();
  Serial.println(waktuakhir);  
  delay(1000);
}
```

justinschoeman, 2018-07-11:
The only thing that jumps out is:
`char lat_buff[9], lon_buff[10];`
Initialisation of this is compiler dependent, so resulting strings may not be null terminated.
Try:
`char lat_buff[9] = {0}, lon_buff[10] = {0};`
?

rogerclarkmelbourne, 2018-07-11:
This sort of thing should be discussed on www.stm32duino.com

closed 2018-07-11 by rogerclarkmelbourne

okumaster, 2018-07-12:
@justinschoeman  But, is no result from the operation, always null