rogerclarkmelbourne/Arduino_STM32 #541: Reseting via DTR not working after flash with changed USART_RX_BUF_SIZE and USART_TX_BUF_SIZE

smarq8, 2018-07-20:
I increase buffers USART_RX_BUF_SIZE and USART_TX_BUF_SIZE to 256 bytes in usart.h file. After I upload it to STM32F1 (bluepill) then reset via DTR not working and I need to press reset button manually each time I upload something.

By the way I am also wondering can I define them in my sketch instead of inside libraries? I also need them in my sketch to know what is actual value but currently I don't know how to properly include it.

I tried every combination, different values and it always stop working when they are both 256.

[image]

stevstrong, 2018-07-23:
It may has to do with the fact that you compile with the option "Fastest (-O3)" and that the ring buffer's head and tail variables are not declared as volatile.
Is this problem happening also when you compile with the default option "Smallest"?

smarq8, 2018-07-23:
It happen for all profiles and almost all values between 230 and 260. Sometime when it wait for reset then green led is fade blinking.
As I watch in files then all variables in rb are volatile but only int F1, F4 are not volatile.
```
F1 version ring_buffer.h:
typedef struct ring_buffer {
    volatile uint8 *buf; /**< Buffer items are stored into */
    volatile uint16 head;         /**< Index of the next item to remove */
    volatile uint16 tail;         /**< Index where the next item will get inserted */
    volatile uint16 size;         /**< Buffer capacity minus one */
} ring_buffer;

F4 version ring_buffer.h:
typedef struct ring_buffer {
    volatile uint8 *buf; /**< Buffer items are stored into */
    uint16 head;         /**< Index of the next item to remove */
    uint16 tail;         /**< Index where the next item will get inserted */
    uint16 size;         /**< Buffer capacity minus one */
} ring_buffer;
```

here are also my hex files form 3 different profiles.
[hw_serial_test.zip](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/2219270/hw_serial_test.zip)