rogerclarkmelbourne/Arduino_STM32 #546: Help! Why the "new" can't be used with the "Bluepill" board? Is there a workaround?

human890209, 2018-07-30:
Hi,
I'm a newbie of "Bluepill" board. I used to use Arduino.
Today I got my "Bluepill" board work with Arduino IDE. I got the Blink example working and start to try other function.
But I've found that Someclass* pointer = new Someclass(); doesn't work. 
I made an example sketch to test:

`
class MYCLASS
{
public:
	MYCLASS(){content = 99;};
	byte content;
};

MYCLASS mc;

MYCLASS* mcp;

void setup()
{
Serial.begin(115200);
pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
	Serial.println("loop");
	digitalWrite(LED_BUILTIN, HIGH);
	delay(500);
	digitalWrite(LED_BUILTIN, LOW);
	delay(500);
	Serial.println(mc.content);
	mcp = new MYCLASS();
	Serial.println((*mcp).content + 1);
}
`

I tried it on my other Arduino, they all work fine. But the "Bluepill" board will freeze when it comes this line: mcp = new MYCLASS();

If I put this "mcp = &mc;", it works. So it is the problem of the "new" word.
Is there a workaround for "new", or should it be another syntax for the "new" word?

stevstrong, 2018-07-30:
https://stackoverflow.com/questions/12248703/creating-an-instance-of-class

human890209, 2018-07-30:
Hi, sevstrong
Thanks for your answer. I googled the "POD-type" and comments out the constructor:
`class MYCLASS
{
public:
	//MYCLASS(){content = 99;};
	byte content = 99;
};

MYCLASS mc;

MYCLASS* mcp;

void setup()
{
Serial.begin(115200);
pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
	Serial.println("loop");
	digitalWrite(LED_BUILTIN, HIGH);
	delay(500);
	digitalWrite(LED_BUILTIN, LOW);
	delay(500);
	Serial.println(mc.content);
	mcp = new MYCLASS();
	Serial.println((*mcp).content + 1);
}` 
And the "Bluepill" board works.
But I'm still confusing. 
I use this Pointer cause I want to make a none-POD-Type instance in the dynamic memory.  I tried it with almost all the Arduino and they all work.
How to perform the "Foo* foo1 = new Foo (); " if Foo() is not a POD-type with the "Bluepill" board?
Why Arduino can do it but "Bluepill" board can't? How to create instances of the none-POD-Type class in dynamic memory with a pointer target them?

human890209, 2018-07-30:
Hi, 
It works in Arduino IDE.
It fails in Sloeber.

closed 2018-07-30 by human890209