rogerclarkmelbourne/Arduino_STM32 #553: Ubuntu 18.04.1 LTS installation problems

Nic30, 2018-08-12:
Hello,
It seems to me that Arduino_STM32 can not detect x64 architecture right.

I just downloaded arduino-1.8.5 (x64).
Downloaded this repo in $HOME/Arduino/hardware
And run $GIT/tools/linux64/install.sh
I have selected the board (STM32F4 discovery)

Now when I try to flash the board, there is an error:

/home/nic30/Arduino/hardware/Arduino_STM32/tools/linux/stlink_upload: line 2: /home/nic30/Arduino/hardware/Arduino_STM32/tools/linux/stlink/st-flash: No such file or directory

File exists but is not possible to excecute it correctly however x64 version is working but it is not used.
```
# cd ~/Arduino/hardware/Arduino_STM32/tools/linux/stlink
# ls -l
total 708
-rwxr-xr-x 1 nic30 nic30 172752 srp 12 14:43 st-flash
-rwxr-xr-x 1 nic30 nic30 170300 srp 12 14:43 st-info
-rwxr-xr-x 1 nic30 nic30 143235 srp 12 14:43 st-term
-rwxr-xr-x 1 nic30 nic30 231592 srp 12 14:43 st-util

# ./st-flash 
bash: ./st-flash: No such file or directory

# cd ~/Arduino/hardware/Arduino_STM32/tools/linux64/stlink
./st-flash 
invalid command line
stlinkv1 command line: ./st-flash [--debug] [--reset] [--serial <iSerial>] {read|write} /dev/sgX path addr <size>
stlinkv1 command line: ./st-flash [--debug] /dev/sgX erase
stlinkv2 command line: ./st-flash [--debug] [--reset] [--serial <iSerial>] {read|write} path addr <size>
stlinkv2 command line: ./st-flash [--debug] [--serial <iSerial>] erase
                       use hex format for addr, <iSerial> and <size>

```

How to force Arduino_STM32 to use linux64 instead linux tools?

Nic30, 2018-08-12:
After changing the paths in STM32F4 platform.txt it is still not working
```
USB Status [unknown]
/home/nic30/Arduino/hardware/Arduino_STM32/tools/linux64/stlink_upload: line 31: 19349 Segmentation fault      (core dumped) $(dirname $0)/stlink/st-flash write "$4" 0x8000000
USB Status [unknown]
Waiting for tty device  

find: ‘/dev/vboxusb’: Permission denied
 should now be available.

```
(I am running my system on real computer not in virtualbox)

Nic30, 2018-08-12:
I have downloaded stlink manually and compiled it and installed it.

Now arduino writes "Done uploading." But there are shill some erros.
```
Archiving built core (caching) in: /tmp/arduino_cache_937427/core/core_Arduino_STM32_STM32F4_discovery_f407_usb_cfg_usb_nc,opt_osstd_5612093573d6c4bf27994df437e799f0.a
Sketch uses 15504 bytes (1%) of program storage space. Maximum is 1048576 bytes.
Global variables use 7336 bytes (5%) of dynamic memory, leaving 123736 bytes for local variables. Maximum is 131072 bytes.
USB Status [unknown]
2018-08-12T15:27:53 INFO usb.c: -- exit_dfu_mode
2018-08-12T15:27:53 INFO common.c: Loading device parameters....
2018-08-12T15:27:53 INFO common.c: Device connected is: F4 device, id 0x20006411
2018-08-12T15:27:53 INFO common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
2018-08-12T15:27:53 INFO common.c: Attempting to write 21036 (0x522c) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Sector:0x0 Size:0x4000 st-flash 1.4.0-47-gae717b9
EraseFlash - Sector:0x1 Size:0x4000 
Flash page at addr: 0x08000000 erased2018-08-12T15:27:53 INFO common.c: Finished erasing 2 pages of 16384 (0x4000) bytes
2018-08-12T15:27:53 INFO common.c: Starting Flash write for F2/F4/L4
2018-08-12T15:27:53 INFO flash_loader.c: Successfully loaded flash loader in sram

Flash page at addr: 0x08004000 erased2018-08-12T15:27:54 INFO common.c: Starting verification of write complete
2018-08-12T15:27:54 INFO common.c: Flash written and verified! jolly good!

enabling 32-bit flash writes
size: 21036
USB Status [unknown]
find: ‘/dev/vboxusb’: Permission denied
Waiting for tty device  

 should now be available.
```

Nic30, 2018-08-12:
When I try to run Burn Bootloader there is an NullPointerException.

```
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:362)
	at processing.app.Editor.lambda$handleBurnBootloader$14(Editor.java:2519)
	at java.lang.Thread.run(Thread.java:748)
Error while burning bootloader.
 should now be available.
```

Nic30, 2018-08-12:
But board seems to be working, except I had to press reset after each flash to run the code.
And Serial is not working.
If I use baudrate 9600 (in arduinio code and in receiver program (picocom)) there are some non ascii chars.
If I use baudrate 115200 there is nothing. I do not have an osciloscope for debug at the moment but
it seems that there is some problem with default clock.

rogerclarkmelbourne, 2018-08-12:
See the Readme.md

> PRIMARY SUPPORT FORUM: http://www.stm32duino.com/

closed 2018-08-12 by rogerclarkmelbourne

thibaudk, 2019-04-05:
hello @Nic30 

I am having the same problem on Ubuntu 18.10, altho I don't seem to have your permission issue.
Did you ever find a solution on the forum ?

Cheers !

Nic30, 2019-04-05:
I do not remember exact steps but I did downloaded new stlink_upload and updated the installation.
It is nearly a year and I do not remember exactly.

thibaudk, 2019-04-05:
ok, thanks!

I'll look into stlink_upload, maybe without the IDE first.
I'll post the results here if they are successful.