rogerclarkmelbourne/Arduino_STM32 #560: GD32F103 Missing

Matheus-Garbelini, 2018-08-29:
Hello, I'm not being able to find the GD32F103 option anymore after a update.
The strange thing is that GD32 is still located in variants the folder. Do you know what's the reason for the GD option to disappear?

[image]

rogerclarkmelbourne, 2018-08-29:
GD32 support was removed ages ago, because the GD32 is not 100% compatible with the STM32.

The folder was retained for reference purposes only.

I will delete it as it’s obviously causing confusion

rogerclarkmelbourne, 2018-08-29:
Look in the commit history for boards.txt

rogerclarkmelbourne, 2018-08-29:
You can try adding the lines from boards.txt into your version of boards.txt

If it works for you, I won't delete the variant folder, but basically no one (me included) is doing any support for the GD32

Matheus-Garbelini, 2018-08-29:
Thank, I found the repository link:
https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/859a447ecfb74a4a2b173d05e8e956e23a8aa05a

closed 2018-08-29 by Matheus-Garbelini

Matheus-Garbelini, 2018-08-30:
@rogerclarkmelbourne interesting, I believe I hadn't have problem yet because lora modules such as the sx1278 are more simple.