rogerclarkmelbourne/Arduino_STM32 #573: Weird passing of command line arguments in "Arduino_STM32/tools/win/maple_upload.bat"

YChvanov, 2018-11-04:
Using "%1 %2 %3 %4 %5 %6 %7 %8 %9" seems strange, shouldn't "%*" be used?

rogerclarkmelbourne, 2019-02-28:
I doubt all the args are even used.

I've already cleaned the STLink upload script because most of the args are unused, I suspect the same applies to this.

stevstrong, 2019-09-16:
Can someone test whether the change suggested by @ultrakiller2 works correctly and does not cause side effects?

rogerclarkmelbourne, 2019-09-16:
I'm not sure what benefit this has.

Playtform.txt doesn't send 9 args to maple_upload.bat, it only sends 5 args.

`tools.maple_upload.upload.pattern="{path}/{cmd}" {serial.port.file} {upload.altID} {upload.usbID} "{build.path}/{build.project_name}.bin"`

AFIK we don't have the source code to the JAR file that handles the upload, and I don't know what would happen if it expects 9 args and the %* only sends 5 because thats what platform.txt sends

If this is really a problem, the best solution is to decompile the maple upload JAR and remove the need for any redundant args.

But since it works OK, it seems a lot of work for zero gain, apart from neatness

stevstrong, 2019-09-16:
Roger, I agree. Should we close then this issue?

rogerclarkmelbourne, 2019-09-16:
Yes. Close it.

Its not a bug.

stevstrong, 2019-09-16:
It's not a bug so I close it.

closed 2019-09-16 by stevstrong