rogerclarkmelbourne/Arduino_STM32 #576: Program and data memory consumtion signnificantly increased after #502

espHorst, 2018-11-21:
I have a small issue with this amazing library. Starting with 

b16686a06ac71f3e636a47af04ed930e13af665a Allow IRQ overriding (which by the way is VERY useful), #502 

the program and data memory consumption of my sketches significantly increased:

b16686a06ac71f3e636a47af04ed930e13af665a 
Sketch uses 61180 bytes (93%) of program storage space. Maximum is 65536 bytes.
Global variables use 18240 bytes (89%) of dynamic memory, leaving 2240 bytes for local variables. Maximum is 20480 bytes.

One commit before:

a5188b8cc3077ad3a979e74653507550d67bb371
Sketch uses 56620 bytes (86%) of program storage space. Maximum is 65536 bytes.
Global variables use 18024 bytes (88%) of dynamic memory, leaving 2456 bytes for local variables. Maximum is 20480 bytes.

The issue seems to be caused by the changes in the isrs.S file.
If I go to b16686a06ac71f3e636a47af04ed930e13af665a and just use the isrs.S
from a5188b8cc3077ad3a979e74653507550d67bb371 there is no increased memory consumption.

espHorst, 2018-11-24:
I did now also do tests with an totally empty sketch, here I see the same behaviour:

b16686a
Sketch uses 13092 bytes (19%) of program storage space. Maximum is 65536 bytes.
Global variables use 2464 bytes (12%) of dynamic memory, leaving 18016 bytes for local variables. Maximum is 20480 bytes.

a5188b8
Sketch uses 7204 bytes (10%) of program storage space. Maximum is 65536 bytes.
Global variables use 1984 bytes (9%) of dynamic memory, leaving 18496 bytes for local variables. Maximum is 20480 bytes.

rogerclarkmelbourne, 2019-01-30:
The consensus from the community was that they want the ability to override the IRQ's so this is a necessary side effect.

closed 2019-01-30 by rogerclarkmelbourne