rogerclarkmelbourne/Arduino_STM32 #577: Timer declaration errors during compile or upload

NicholasTracy, 2018-11-27:
Fresh install of the library in Arduino 1.8.7 Using the blue pill 128k STM32F103C board and an ST-Link V2 clone with recent firmware update.

I get this error on compile of every example provided so far as well as Arduino examples like the simple blink code.

Not sure if this is something I need to worry about or not. some examples compile and run fine and others wont run or flash to the target.

```
P:\Arduino\Sketches\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/timer.h:150:18: warning: type of 'timer4' does not match original declaration [enabled by default]

 extern timer_dev timer4;

                  ^

P:\Arduino\Sketches\hardware\Arduino_STM32\STM32F1\cores\maple\libmaple\timer.c:67:11: note: previously declared here

 timer_dev timer4 = GENERAL_TIMER(4);

           ^

P:\Arduino\Sketches\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/timer.h:138:18: warning: type of 'timer1' does not match original declaration [enabled by default]

 extern timer_dev timer1;

                  ^

P:\Arduino\Sketches\hardware\Arduino_STM32\STM32F1\cores\maple\libmaple\timer.c:52:11: note: previously declared here

 timer_dev timer1 = ADVANCED_TIMER(1);

           ^

P:\Arduino\Sketches\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/timer.h:146:18: warning: type of 'timer3' does not match original declaration [enabled by default]

 extern timer_dev timer3;

                  ^

P:\Arduino\Sketches\hardware\Arduino_STM32\STM32F1\cores\maple\libmaple\timer.c:62:11: note: previously declared here

 timer_dev timer3 = GENERAL_TIMER(3);

           ^

P:\Arduino\Sketches\hardware\Arduino_STM32\STM32F1\system/libmaple/stm32f1/include/series/adc.h:45:16: warning: type of 'adc1' does not match original declaration [enabled by default]

 extern adc_dev adc1;

                ^

P:\Arduino\Sketches\hardware\Arduino_STM32\STM32F1\cores\maple\libmaple\adc_f1.c:43:9: note: previously declared here

 adc_dev adc1 = {

         ^

P:\Arduino\Sketches\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/timer.h:142:18: warning: type of 'timer2' does not match original declaration [enabled by default]

 extern timer_dev timer2;

                  ^

P:\Arduino\Sketches\hardware\Arduino_STM32\STM32F1\cores\maple\libmaple\timer.c:57:11: note: previously declared here

 timer_dev timer2 = GENERAL_TIMER(2);

           ^
```

NicholasTracy, 2018-11-28:
Its not just one example code that produces the error. Any example that uses timers seems to do this. But the main one that I was compiling to run on the Blue Pill STM32F103C is this one provided in the STM32 specific examples folder;

```
/*
  Blink without delay

  Turns on and off the built-in light emitting diode (LED), without
  using the delay() function.  This means that other code can run at
  the same time without being interrupted by the LED code.

  created 2005
  by David A. Mellis
  modified 17 Jun 2009
  by Tom Igoe
  modified for Maple 27 May 2011
  by Marti Bolivar
*/

// Variables:
int previousMillis = 0;        // will store the last time the LED was updated
int interval = 1000;            // interval at which to blink (in milliseconds)

void setup() {
    // Set up the built-in LED pin as output:
    pinMode(PC13, OUTPUT);
}

void loop() {
    // Check to see if it's time to blink the LED; that is, if the
    // difference between the current time and last time we blinked
    // the LED is bigger than the interval at which we want to blink
    // the LED.
    if (millis() - previousMillis > interval) {
        // Save the last time you blinked the LED
        previousMillis = millis();

        // If the LED is off, turn it on, and vice-versa:
        digitalWrite(PC13,!digitalRead(PC13));// Turn the LED from off to on, or on to off
    }
}
```

stevstrong, 2018-11-28:
I think you did not follow exactly the intallatino procedure.
The core should go directly under Arduino folder.
And not on a network drive (P).

rogerclarkmelbourne, 2018-11-28:
I just installed Arduino 1.8.7, just in case they had changed something, and its all working fine, including the code posted above.

This is not a bug. 

You can't have installed things correctly

This is not a support forum, please see the FAQ on www.stm32duino.com

NicholasTracy, 2018-11-30:
I understand this is not a support forum but a bug tracking section is it not?
As such I posted here because I was experiencing a bug.
Before you start assuming that this is user error and not a bug please try to be a little more open minded and comprehensive. I notice that you seem to be quick to discount a lot of issues on here as support help issues.

Lets get back to the issues at hand. I have installed the core directly. I have never had an issue with using my network drive as the primary directory for my Arduino installation, I have done this for portability. The way Arduino loads the files from the directory is the same either way. But to be fair I went ahead and installed a clean and recent Arduino install, using the default directories and installed the core as directed. The error still appears. 

Thanks for your time.

stevstrong, 2018-11-30:
It is not about Arduino install. The path where the Arduino_STM32 files are located is wrong, not conforming the installation procedure.
http://wiki.stm32duino.com/index.php?title=Installation
And this is not a bug, no other user complains.

NicholasTracy, 2018-11-30:
Curious that you say that. How is it not installed per the instructions?
"[Arduino sketches folder]/[hardware]/[Arduino_STM32]" 
My Arduino sketch folder is located on my P drive @ P:\Arduino\Sketches\hardware\Arduino_STM32 which is correct according to your link. (My Arduino installation has all its main directories pointing to this folder. otherwise the core wouldn't show up in my boards list at all.)

I find it odd that you think having the sketch folder on a different drive than the Arduino installation would make any difference. I have proven that to be incorrectly assumed already though by doing a fresh install of arduino and using the default directories on the C:/ drive. I installed the arduino sam boards package and then unzipped the downloaded files correctly to the hardware folder per your instructions. (I'm not new to this, I have a dozen other cores installed that I use the same way, this is the only one giving me a hard time).

So explain to me again how I'm doing some thing wrong?
Also, just because I'm the only user complaining at this time about this doesn't mean it isn't a bug.
I've been a software developer long enough to know that just isn't how things work.
All I'm asking is that you guys try a little harder than just telling me I'm installing it wrong, did you even look at the errors?

JiriBilek, 2018-11-30:
@NicholasTracy:
Although the installation procedure requires to put the Arduino_STM32 folder in the sketch folder, I put the Arduino_STM32 folder directly in the arduino-1.8.7\hardware folder (which resides on drive D:). I am running it successfully for a long time, no such problems as you have.
Maybe you should try it.

[image]

NicholasTracy, 2018-11-30:
@JiriBilek 
Thanks, That was somewhat helpful, Out of curiosity I tried another clean install with that location instead of the Sketches directory. The errors are still there. 
More info that may or may not be useful.
I am trying to compile using these options:
Board: Generic STM32F103C
Variant: STM33F103CB (20K RAM. 128K Flash)
Upload Method: ST Link
CPU Speed 72mhz
Optimize: Smallest Code with LTO

NicholasTracy, 2018-11-30:
Update, The errors go away when I turn off LTO.

JiriBilek, 2018-11-30:
Yes, just now I realized it, too :)

NicholasTracy, 2018-11-30:
I updated the title. Please confirm that I am not crazy and that LTO is not functioning properly?

NicholasTracy, 2018-11-30:
Whichever is shipped with Arduino 1.8.7 I haven't messed with that.

JiriBilek, 2018-11-30:
I can reproduce it with arm-none-eabi-gcc\4.8.3-2014q1

fpistm, 2018-11-30:
Ok. Thanks @JiriBilek . I already met issue with LTO enabled with some version of arm toolchain. Don't know about this one (4.8.3 is the default one from arduino).
Just one other remarks, your log show only warning, not error...

NicholasTracy, 2018-11-30:
1.8.7 appears to be using 4.8.3.
I should have done it from the start but I just turned on verbose logging during compile. LTO is the culprit.
 
Complete output using LTO with the Blink without Delay sketch provided with the core.
```
D:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Nick\AppData\Local\Arduino15\packages -hardware P:\Arduino\Sketches\hardware -tools D:\Program Files (x86)\Arduino\tools-builder -tools D:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Nick\AppData\Local\Arduino15\packages -built-in-libraries D:\Program Files (x86)\Arduino\libraries -libraries P:\Arduino\Sketches\libraries -fqbn=Arduino_STM32:STM32F1:genericSTM32F103C:device_variant=STM32F103CB,upload_method=STLinkMethod,cpu_speed=speed_72mhz,opt=oslto -ide-version=10807 -build-path C:\Users\Nick\AppData\Local\Temp\arduino_build_545889 -warnings=all -build-cache C:\Users\Nick\AppData\Local\Temp\arduino_cache_702340 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Nick\AppData\Local\Temp\arduino_modified_sketch_536697\BlinkWithoutDelay.ino
D:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware D:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Nick\AppData\Local\Arduino15\packages -hardware P:\Arduino\Sketches\hardware -tools D:\Program Files (x86)\Arduino\tools-builder -tools D:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Nick\AppData\Local\Arduino15\packages -built-in-libraries D:\Program Files (x86)\Arduino\libraries -libraries P:\Arduino\Sketches\libraries -fqbn=Arduino_STM32:STM32F1:genericSTM32F103C:device_variant=STM32F103CB,upload_method=STLinkMethod,cpu_speed=speed_72mhz,opt=oslto -ide-version=10807 -build-path C:\Users\Nick\AppData\Local\Temp\arduino_build_545889 -warnings=all -build-cache C:\Users\Nick\AppData\Local\Temp\arduino_cache_702340 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Nick\AppData\Local\Temp\arduino_modified_sketch_536697\BlinkWithoutDelay.ino
Using board 'genericSTM32F103C' from platform in folder: D:\Program Files (x86)\Arduino\hardware\Arduino_STM32\STM32F1
Using core 'maple' from platform in folder: D:\Program Files (x86)\Arduino\hardware\Arduino_STM32\STM32F1
Build options changed, rebuilding all
Detecting libraries used...
"C:\\Users\\Nick\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -flto -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10807 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/include" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/stm32f1/include" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/stm32f1" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/usb_lib" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\cores\\maple" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\variants\\generic_stm32f103c" "C:\\Users\\Nick\\AppData\\Local\\Temp\\arduino_build_545889\\sketch\\BlinkWithoutDelay.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\Nick\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -flto -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10807 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/include" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/stm32f1/include" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/stm32f1" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/usb_lib" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\cores\\maple" "-ID:\\Program Files (x86)\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\variants\\generic_stm32f103c" "C:\\Users\\Nick\\AppData\\Local\\Temp\\arduino_build_545889\\sketch\\BlinkWithoutDelay.ino.cpp" -o "C:\\Users\\Nick\\AppData\\Local\\Temp\\arduino_build_545889\\preproc\\ctags_target_for_gcc_minus_e [... cut ...]
Connection mode : Normal.
Device ID:0x410 
Device flash Size : 128 Kbytes
Device family :STM32F10xx Medium-density

Loading file...
Flash Programming:
  File : C:\Users\Nick\AppData\Local\Temp\arduino_build_545889\BlinkWithoutDelay.ino.bin
  Address : 0x08000000
Flash memory programming...
±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±± 0%
 0%ÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛ 57%ÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛÛ 100%
Flash memory programmed in 1s and 62ms.
Programming Complete.

MCU Reset.

Application started.
```

NicholasTracy, 2018-11-30:
@fpistm 
Ok, in some examples the example sketches will run fine on the target. in others they do not run or they hang. When turning off LTO they work normally. 
I changed my verbiage in the title. My apologies.

rogerclarkmelbourne, 2018-11-30:
LTO has never been very beneficial.

I've removed those options in #c7054992554b27ec6bdfc016347806454ab69815

fpistm, 2018-11-30:
Ok that's what I think.
Fix the warning could be fine.
Anyway, I really think the issue can come from the arm gcc version.
4.8.3 is old and LTO was very buggy as fairly new in 4.8 (gnu ld 2.21 I think)

rogerclarkmelbourne, 2018-11-30:
@fpistm 

I don't think LTO actually resulted in the code being much smaller. 

For most people its not worth using, and has always potentially caused runtime issues due to the level of optimisation.

I did ask the community if they would accept changing the default optimisation to either -O1 -O2 or -O3 but most people voted to remain using -Os as the default since it seemed to be the most reliable, and some people were worried about the code size when using the other optimisation settings.

Personally, since the F103C8 has such high yields in its production, 99% of the devices sold as F103C8 are actually F103CB  and have 128k flash, so for hobbyists who  are happy to test the devices individually, they can make use of the free 64k extra if they they don't have enough flash space for their project

NicholasTracy, 2018-11-30:
I would like to vouch for LTO in cases where users like myself are writing lengthy bits of code with multiple libraries, one of the main reasons I am exploring STM32 as a cheap alternative to others. I have seen significant improvements in most of my code. But on STM32103 boards using -0 does it really improve anything? I know for other boards it usually does.

fpistm, 2018-11-30:
LTO can have some benefits on large programs. For blink this will not result on smaller code.
I didn't advise LTO to be used. It's up to end user to use it or not for his use case.

NicholasTracy, 2018-11-30:
@fpistm 
Ok, noted.
I have yet to start writing anything significant for this target. most of what I have written has been optimized to fit on most of the 8-bit chips from Micro using the MCUdude cores and some of the ESP targets and LTO has been beneficial in those cases. Is this something that could be made to work? or is this a bug in gcc 4.8.3?

rogerclarkmelbourne, 2018-11-30:
@NicholasTracy 

You should probably use the official STM Arduino core which @fpistm administers.

See https://github.com/stm32duino

Not is old core.

However I'm not sure if the official core supports optimisation mode selection

NicholasTracy, 2018-11-30:
I don't think it does. I actually just loaded it up to give it a try.

Thanks for looking in to this.

JiriBilek, 2018-11-30:
@fpistm, this unfortunately introduces a new problem: zillions of errors
`C:\Users\Jiri\AppData\Local\Arduino15\packages\STM32\tools\arm-none-eabi-gcc\6-2017-q2-update/bin/arm-none-eabi-ar: C:\Users\Jiri\AppData\Local\Temp\arduino_build_452231\core\HardwareSerial.cpp.o: plugin needed to handle lto object`

fpistm, 2018-11-30:
Ok, this would required more stuff to perform this.
LTO is not easy to use...

rogerclarkmelbourne, 2019-01-28:
I removed the LTO options some time ago, because they caused all sorts of problems

closed 2019-01-28 by rogerclarkmelbourne